furkantokac/buildroot #1: Enable dmesg messages and USB

ricardojlrufino, 2020-03-31:
Olá, fiz o teste da sua imagem e funciona correamente parabéns.

Estou tentando habilitar alguns recursos aos poucos, e estou tendo dificudades. Não consegui fazer os logs funcionarem dai fica bem dificil debuger.

Pode me ajudar ?

furkantokac, 2020-03-31:
Sure. 

You can find the disabled kernel configurations from [here](https://github.com/furkantokac/buildroot/blob/ftdev/board/ftdev/rpi3/docs/distro_optimization/fcond04/README.config).

For more detailed info, I suggest you to compare original kernel defconfig (which is bcm2709_defconfig) and my fastboot kernel defconfig[1]. You can find the original kernel defconfig from reference [2], but it doesn't have same styling with my kernel defconfig so it is hard to compare side by side. To obtain an original kernel defconfig with the same style of my kernel defconfig, just compile my fastboot defconfig[3] with the parameter `BR2_LINUX_KERNEL_DEFCONFIG="bcm2709"` instead of `BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="board/ftdev/rpi3/custom_files/kernel-fastboot.config"`, then save linux defconfig and compare them easily.

**References**
[1] fastboot kernel defconfig
https://github.com/furkantokac/buildroot/blob/ftdev/board/ftdev/rpi3/custom_files/kernel-fastboot.config

[2] original kernel defconfig
https://github.com/raspberrypi/linux/blob/c117a8bccf37bfba323065b566cf999ed4629a4a/arch/arm/configs/bcm2709_defconfig

[3] fastboot defconfig
https://github.com/furkantokac/buildroot/blob/ftdev/configs/ftdev_rpi3_fastboot_defconfig

ricardojlrufino, 2020-03-31:
I don't know if I followed the instructions correctly, but in the test I did, I didn't find the differences I was looking for (related to debug log and USB)
My changes:
diff configs/ftdev_rpi3_fastboot_defconfig configs/ftdev_rpi3_fastboot_v2_defconfig 
```
16,17c16
< BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG=y
< BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="board/ftdev/rpi3/custom_files/kernel-fastboot.config"
---
> BR2_LINUX_KERNEL_DEFCONFIG="bcm2709"
```

I saw that the .config of the root folder has "no differences".
As I understand it the difference will be in, [1], right ?!:
[1] output/build/linux-c117a8bccf37bfba323065b566cf999ed4629a4a/.config

Comparing the before and after, USB remains disabled
[image]

I took the settings below from 'raspberrypi3_defconfig' to test
diff configs/ftdev_rpi3_fastboot_defconfig configs/ftdev_rpi3_fastboot_**v3**_defconfig 
```
13,17c13,15
< BR2_LINUX_KERNEL_CUSTOM_GIT=y
< BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/raspberrypi/linux.git"
< BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="c117a8bccf37bfba323065b566cf999ed4629a4a"
< BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG=y
< BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="board/ftdev/rpi3/custom_files/kernel-fastboot.config"
---
> BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
> BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="$(call github,raspberrypi,linux,83b36f98e1a48d143f0b466fcf9f8c4e382c9a1c)/linux-83b36f98e1a48d143f0b466fcf9f8c4e382c9a1c.tar.gz"
> BR2_LINUX_KERNEL_DEFCONFIG="bcm2709
```

This make impossible to compare like you said

ricardojlrufino, 2020-03-31:
running make defconfig-save in linux kernel folder proved the compare results, but 
a series of new questions were asked, which I did not know how to answer ... = /

ricardojlrufino, 2020-04-01:
I was able to enable logs, thanks to your help.
Thank you very much

I found this error enabling QT
```
>>> qt5declarative 5.11.3 Installing to target
cp -dpf /opt/builds/buildroot-rasp3/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/bin/qml* /opt/builds/buildroot-rasp3/output/target/usr/bin
cp -dpfr /opt/builds/buildroot-rasp3/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/qml /opt/builds/buildroot-rasp3/output/target/usr
cp -dpf /opt/builds/buildroot-rasp3/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/libQt5Qml*.so.* /opt/builds/buildroot-rasp3/output/target/usr/lib
cp -dpfr /opt/builds/buildroot-rasp3/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/qt/plugins/qml* /opt/builds/buildroot-rasp3/output/target/usr/lib/qt/plugins
cp -dpf /opt/builds/buildroot-rasp3/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/libQt5Quick*.so.* /opt/builds/buildroot-rasp3/output/target/usr/lib
cp: não foi possível obter estado de '/opt/builds/buildroot-rasp3/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/libQt5Quick*.so.*': Arquivo ou diretório inexistente
package/pkg-generic.mk:331: recipe for target '/opt/builds/buildroot-rasp3/output/build/qt5declarative-5.11.3/.stamp_target_installed' failed
make: *** [/opt/builds/buildroot-rasp3/output/build/qt5declarative-5.11.3/.stamp_target_installed] Error 1
```

tried with version 5.6, but also got errors ..

ricardojlrufino, 2020-04-01:
after a rebuild its works
- delete qt5 folders
- make qt5declarative-reconfigure

furkantokac, 2020-04-01:
I highly recommend you to compile qt-everywhere yourself statically, independent from the Buildroot. I share my script that does the job. Just compile your buildroot, put the script in the same directory that has your compiled "buildroot" directory, change the Qt version however you want and run the script it. It will do everything for you.

```sh
#!/bin/bash

SCRIPT_VERSION="0.2"
BASE_PATH=`pwd`

###
### Qt Version Params
###
QT_MAIN_VERSION="5.12"
QT_SUB_VERSION="6"
QT_FULL_VERSION=$QT_MAIN_VERSION.$QT_SUB_VERSION

###
### Qt Configure Params
###
FT_DEVICE=linux-rasp-pi3-g++
FT_SYSROOT=$BASE_PATH/buildroot/output/host/usr/arm-buildroot-linux-gnueabihf/sysroot
FT_CROSS_COMPILE=$BASE_PATH/buildroot/output/host/usr/bin/arm-buildroot-linux-gnueabihf-
FT_PREFIX=/usr/lib/qt5      # Prefix that will be for target device
FT_HOSTPREFIX=../aaaout/    # Compile to where on host device
FT_EXTPREFIX=$FT_HOSTPREFIX

###
### Fire up
###
mkdir gqt
cd gqt

wget https://download.qt.io/archive/qt/$QT_MAIN_VERSION/$QT_FULL_VERSION/single/qt-everywhere-src-$QT_FULL_VERSION.tar.xz
tar xJf qt-everywhere-src-$QT_FULL_VERSION.tar.xz
cd qt-everywhere-src-$QT_FULL_VERSION

mkdir aaabuild aaaout
cd aaabuild

./configure -release -static -opensource -confirm-license -v \
            -prefix $FT_PREFIX -hostprefix $FT_HOSTPREFIX -sysroot $FT_SYSROOT -extprefix $FT_EXTPREFIX \
            -device $FT_DEVICE -device-option CROSS_COMPILE=$FT_CROSS_COMPILE \
            -opengl es2 -make libs \
            -no-sql-mysql -no-sql-psql -no-sql-sqlite -no-xcb -nomake tests -nomake examples -no-use-gold-linker \
            2>&1 | tee configure.log

make -j8
make install

echo -e "\n#######################DONE##############################################"
echo ">>> Your QMake is ready on './qt-everywhere-src-$QT_FULL_VERSION/aaaout/bin/qmake'"
echo -e "#######################DONE##############################################\n"
```

furkantokac, 2020-04-01:
If you got any problem about Qt, please open a new issue.

closed 2020-04-01 by furkantokac

ricardojlrufino, 2020-04-03:
amazing, thanks, I will test

furkantokac, 2020-04-04:
No problem. It will be great if you can report the result to this issue : https://github.com/furkantokac/buildroot/issues/2

magnock, 2020-05-28:
Thanks @furkantokac for all theses details.
@ricardojlrufino, can you please explain how did you managed to enable USB ? I need to do the same for Network / WLAN. What config did you add to kernel-fastboot.config ? Thx

furkantokac, 2020-05-28:
@magnock USB and network is disabled from the kernel. You can just enable them by `make linux-xconfig`. See what options are disabled from here : https://github.com/furkantokac/buildroot/tree/ftdev/board/ftdev/rpi3/docs/distro_optimization/fcond04

magnock, 2020-05-29:
Thanks @furkantokac. 
If I run “make linux-xconfig”, I‘ll get all other default stuff in kernel. I just need to patch your kernel-fastboot.config with USB and NETWORK.

If I add "CONFIG_NET=y" and "CONFIG_USB_SUPPORT=y" to kernel-fastboot.config and run “make ftdev_rpi3_fastboot_defconfig”, do I expect to have your 1.75s booting img with USB and NETWORK ? (it will be more that 1.75, I know :) )

furkantokac, 2020-05-30:
@magnock

> If I run “make linux-xconfig”, I‘ll get all other default stuff in kernel. I just need to patch your kernel-fastboot.config with USB and NETWORK.

1. make ftdev_rpi3_fastboot_defconfig
2. make linux-xconfig

You should make the ftdev_rpi3_fastboot_defconfig first, then when you make the linux-xconfig, you'll see the settings of ftdev_rpi3_fastboot_defconfig's kernel.

> If I add "CONFIG_NET=y" and "CONFIG_USB_SUPPORT=y" to kernel-fastboot.config

I don't really recommend directly editing the kernel-fastboot.config since some configurations depens on each other.

> do I expect to have your 1.75s booting img with USB and NETWORK ?

The short answer is yes, if you just enable them without doing any extra work, they should slowdown the boot process. But you can optimize them and go below the 1.75sec. There is no limit of optimization :)

Network connection is directly related to userspace like you need some work for network connection after the kernel loaded so even if you have support them from the kernel, you have to optimize the network connection process. For example, the easiest thing you can do is delaying the connection process. It appearently requires some work if your issue is fastboot.

For the USB connection, kind of same. Like external devices always requires some kind of polling, and if the issue is fastboot, it requires some work according to your requirements. For example, my requirement is actually running a Qt application that reads data from UART so till the Qt application runs and I receive the UART data, I don't consider the boot process is finished. If this requirement includes USB device connection and network connection, it requires another work.

If you have any question, please don't hesitate to ask.

sbartolett, 2021-05-10:
Has anyone had success enabling USB devices?  I've tried everything suggested above without any success so far.

furkantokac, 2021-05-17:
Hey @sbartolett,

Please try to enable disabled things that is related to your use case one by one. 

```
NET
SOUND
HW_RANDOM           # 0.7sn
ALLOW_DEV_COREDUMP  # 0.2sn (Core Release: 2.80a)
STRICT_KERNEL_RWX   #===\ 0.1sn
STRICT_MODULE_RWX   #===/
NAMESPACES          # 0.1sn
FTRACE              # 0.5sn

# Disable USB support
USB_SUPPORT

# Disable debugging
BLK_DEBUG_FS
DEBUG_BUGVERBOSE
DEBUG_FS
DEBUG_MEMORY_INIT
SLUB_DEBUG
PRINTK
BUG
DM_DELAY
ELF_CORE
KGDB
PRINT_QUOTA_WARNING
AUTOFS4_FS

# Followings are mostly affects the size
MEDIA_DIGITAL_TV_SUPPORT
MEDIA_ANALOG_TV_SUPPORT
MEDIA_CAMERA_SUPPORT
MEDIA_RADIO_SUPPORT
INPUT_MOUSEDEV
INPUT_JOYDEV
INPUT_JOYSTICK
INPUT_TABLET
INPUT_TOUCHSCREEN
IIO
RC_CORE
HID_LOGITECH
HID_MAGICMOUSE
HID_A4TECH
HID_ACRUX
HID_APPLE
HID_ASUS
```

It looks like the paremeters that are starting with `INPUT_` are related to your use case. Also, after you enable the keyboard support, disable the autostarting application. Check this: https://github.com/furkantokac/buildroot/issues/14#issuecomment-759085963

sbartolett, 2021-05-17:
Thanks Furkan!  I was able to get this working last week.  I should have updated my comment.  I needed to enable INPUT_* and HID_*.  Probably enabled more than I needed.  I plan on going back and seeing which ones I did need.  Once I do that I will post my results

furkantokac, 2021-05-17:
I'm happy for you! That will be good, it'll be useful to people.

a-matos00, 2022-01-11:
@sbartolett could you share the config in which you managed to enable USB support?

guinmoon, 2025-02-17:
Hi. Tried all the above options. But the USB is still not even powered. Can anyone tell me what the problem is or can you download **kernel-fastboot.config** with USB support? Thanks in advance!