furkantokac/buildroot #3: QT Support - Compile static Apps

ricardojlrufino, 2020-04-04:
I compiled QT using you provided script.
But when I compile the executable it looks like a dynamic:
openglwindow
```
/ mnt / nfs-shared / root / openglwindow: ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU / Linux), dynamically linked, interpreter / lib / ld-, for GNU / Linux 4.14.0, not stripped
```

i need change some config ?

ricardojlrufino, 2020-04-04:
Running on rasp; a got this error
```
QIconvCodec::convertToUnicode: using Latin-1 for conversion, iconv_open failed
QIconvCodec::convertFromUnicode: using Latin-1 for conversion, iconv_open failed
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.egldeviceintegration: Failed to load EGL device integration "eglfs_brcm"
EGL library doesn't support Emulator extensions
 ```

furkantokac, 2020-04-04:
You should include the necessary Qt plugins inside your Qt project since we have no Qt libraries in the RPI. To do this, put all the necessary Qt plugins into the .pro file of your Qt project. For example, to include eglfs plugin, put the following to your .pro file and compile your project again : 
`QTPLUGIN.platforms += qeglfs`

To see the available plugins, check the `aaaout/plugins` folder. Also it is important to keep track of configure log to see if there is something wrong about your Qt compilation. You can find the configure log from `aaabuild/configure.log`, the important line is the last lines which starts with `Configure summary:`.

Last thing, when you're running the Qt application on RPI, don't forget to specify the platform like
`./myQtApp --platform eglfs`

Let me know if you have any problem.

ricardojlrufino, 2020-04-05:
Nothing, i compiled changing:
QT_MAIN_VERSION="5.12"
QT_SUB_VERSION="5"
FT_DEVICE=linux-rasp-pi-g++
Following tips from:  
https://mechatronicsblog.com/cross-compile-and-deploy-qt-5-12-for-raspberry-pi/

I tried using the image 2020-02-13-raspbian-buster-lite.img and my custom image.
Same error: 
```
QIconvCodec::convertToUnicode: using Latin-1 for conversion, iconv_open failed
QIconvCodec::convertFromUnicode: using Latin-1 for conversion, iconv_open failed
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.egldeviceintegration: EGL device integration plugin keys: ("eglfs_emu", "eglfs_kms")
qt.qpa.egldeviceintegration: EGL device integration plugin keys (sorted): ("eglfs_brcm", "eglfs_kms", "eglfs_emu")
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_brcm"
qt.qpa.egldeviceintegration: Failed to load EGL device integration "eglfs_brcm"
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_kms"
qt.qpa.eglfs.kms: New DRM/KMS via GBM integration created
qt.qpa.egldeviceintegration: Using EGL device integration "eglfs_kms"
qt.qpa.eglfs.kms: platformInit: Opening DRM device
qt.qpa.input: static device discovery for type QFlags<QDeviceDiscovery::QDeviceType>(Device_DRM|Device_VideoMask)
qt.qpa.input: Found matching devices ()
qt.qpa.eglfs.kms: Found the following video devices: ()
Could not find DRM device!
Aborted
```

ricardojlrufino, 2020-04-05:
My config.log
```
Building on: linux-g++ (x86_64, CPU features: mmx sse sse2)
Building for: devices/linux-rasp-pi-g++ (arm, CPU features: <none>)
Target compiler: gcc 7.4.0
Configuration: cross_compile compile_examples enable_new_dtags largefile precompile_header release c++11 c++14 c++1z concurrent dbus reduce_exports static stl
Build options:
  Mode ................................... release
  Optimize release build for size ........ no
  Building shared libraries .............. no
  Using C standard ....................... C11
  Using C++ standard ..................... C++1z
  Using ccache ........................... no
  Using gold linker ...................... no
  Using new DTAGS ........................ yes
  Using precompiled headers .............. yes
  Using LTCG ............................. no
  Target compiler supports:
    NEON ................................. no
  Build parts ............................ libs
Qt modules and options:
  Qt Concurrent .......................... yes
  Qt D-Bus ............................... yes
  Qt D-Bus directly linked to libdbus .... no
  Qt Gui ................................. yes
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. yes
  Qt Widgets ............................. yes
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  udev ................................... no
  Using system zlib ...................... yes
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ no
  GLib ................................... no
  iconv .................................. yes
  ICU .................................... no
  Tracing backend ........................ <none>
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  Using system PCRE2 ..................... no
Qt Network:
  getifaddrs() ........................... yes
  IPv6 ifname ............................ yes
  libproxy ............................... no
  Linux AF_NETLINK ....................... yes
  OpenSSL ................................ no
    Qt directly linked to OpenSSL ........ no
  OpenSSL 1.1 ............................ no
  DTLS ................................... no
  SCTP ................................... no
  Use system proxies ..................... yes
Qt Gui:
  Accessibility .......................... yes
  FreeType ............................... yes
    Using system FreeType ................ no
  HarfBuzz ............................... yes
    Using system HarfBuzz ................ no
  Fontconfig ............................. no
  Image formats:
    GIF .................................. yes
    ICO .................................. yes
    JPEG ................................. yes
      Using system libjpeg ............... yes
    PNG .................................. yes
[... 170 lines cut ...]
    jsoncpp .............................. no
    protobuf ............................. no
    libxml2 and libxslt .................. no
    lcms2 ................................ no
    png .................................. yes
    JPEG ................................. yes
    harfbuzz ............................. no
    freetype ............................. no
  Required system libraries:
    fontconfig ........................... no
    dbus ................................. no
    nss .................................. no
    khr .................................. yes
    glibc ................................ yes
  Required system libraries for qpa-xcb:
    x11 .................................. no
    libdrm ............................... yes
    xcomposite ........................... no
    xcursor .............................. no
    xi ................................... no
    xtst ................................. no

```
This can be a problem , right ?
** EGLFS Raspberry Pi ................... no **

ricardojlrufino, 2020-04-05:
/opt/vc/src/hello_pi/hello_triangle (from raspbian-buster-lite.img), example its working

he seems to use EGL

```
#include "bcm_host.h"
#include "GLES/gl.h"
#include "EGL/egl.h"
#include "EGL/eglext.h"

```

furkantokac, 2020-04-05:
Do you have 

> Nothing, i compiled changing:
> QT_MAIN_VERSION="5.12"
> QT_SUB_VERSION="5"
> FT_DEVICE=linux-rasp-pi-g++
> Following tips from:
> ```

It should be `FT_DEVICE=linux-rasp-pi3-g++` instead of `FT_DEVICE=linux-rasp-pi-g++`.

ricardojlrufino, 2020-04-05:
I'm getting, in configure.log

```
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    EGLFS EGLDevice ...................... no
    EGLFS GBM ............................ yes
    EGLFS VSP2 ........................... no
    EGLFS Mali ........................... no
    EGLFS Raspberry Pi ................... no <<< Its right ??
    EGLFS X11 ............................ no
  LinuxFB ................................ yes
  VNC .................................... yes
  Mir client ............................. no
```

Searching for EGL its exists :

```
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/build/lib/libEGL_static.a
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/build/lib/libEGL.so
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/build/lib/libbrcmEGL.so
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/interface/khronos/include/EGL
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/interface/khronos/CMakeFiles/EGL.dir
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/interface/khronos/CMakeFiles/EGL_static.dir
./output/build/rpi-userland-e5803f2c986cbf8c919c60278b3231dcdf4271a6/interface/khronos/CMakeFiles/brcmEGL.dir
./output/target/usr/lib/libEGL.so
./output/target/usr/lib/libbrcmEGL.so
./output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/EGL
./output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/libEGL_static.a
./output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/libEGL.so
./output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/libbrcmEGL.so
```

I copy the libraries to raspberry:
scp /opt/builds/buildroot/output/target/usr/lib/* pi@192.168.15.21:/usr/lib/qt5/

Link output:
root@raspberrypi:/home/pi# ldd openglwindow 
```
        linux-vdso.so.1 (0x7ed9f000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76ec5000)
        libjpeg.so.8 => /usr/lib/qt5/libjpeg.so.8 (0x76e66000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76e53000)
        libgbm.so.1 => /usr/lib/qt5/libgbm.so.1 (0x76e36000)
        libdrm.so.2 => /usr/lib/qt5/libdrm.so.2 (0x76e17000)
        libEGL.so => /usr/lib/qt5/libEGL.so (0x76ddf000)
        libvchostif.so => /usr/lib/qt5/libvchostif.so (0x76db2000)
        libpng16.so.16 => /usr/lib/qt5/libpng16.so.16 (0x76d6a000)
        libz.so.1 => /usr/lib/qt5/libz.so.1 (0x76d3d000)
        libGLESv2.so => /usr/lib/qt5/libGLESv2.so (0x76d19000)
        libbcm_host.so => /usr/lib/qt5/libbcm_host.so (0x76d04000)
        libvcos.so => /usr/lib/qt5/libvcos.so (0x76cea000)
        libvchiq_arm.so => /usr/lib/qt5/libvchiq_arm.so (0x76cd3000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ca9000)
        libstdc++.so.6 => /usr/lib/qt5/libstdc++.so.6 (0x76b55000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76ad3000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76aa6000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76958000)
        /lib/ld-linux-armhf.so.3 (0x76eda000)
        libexpat.so.1 => /usr/lib/qt5/libexpat.so.1 (0x76917000)
        libkhrn_client.so => /usr/lib/qt5/libkhrn_client.so (0x76903000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x768ec000)
        libvcfiled_check.so => /usr/lib/qt5/libvcfiled_check.so (0x768da000)
```

ricardojlrufino, 2020-04-05:
I need load modules ? 
# lsmod
Module                  Size  Used by    Not tainted
ipv6                  434176 18 [permanent]

furkantokac, 2020-04-05:
Can you please share your .pro file ?

ricardojlrufino, 2020-04-05:
Can you send a binary file file ?

ricardojlrufino, 2020-04-05:
I cleaned everything

furkantokac, 2020-04-06:
Let me check. Give me sometime please.

furkantokac, 2020-04-07:
I saw that 5.12.6 has problem, try 5.12.1. I'll share an example binary soon.

Also I can confirm that, the configure.log should have include the following log :
`EGLFS Raspberry Pi ................... yes`.

ricardojlrufino, 2020-04-10:
Can you send me some binary to test before ... please?!?!
I wasted a lot of time trying to compile, not knowing if it was working

furkantokac, 2020-04-14:
This works for me : https://github.com/furkantokac/buildroot/blob/ftdev/board/ftdev/rpi3/rootfs_overlay/root/QmlTestApplication

Download the file and run it as : `/root/QmlTestApplication --platform eglfs`

This is configuration I use : https://github.com/furkantokac/buildroot/blob/ftdev/build-rpi3-qt.sh

furkantokac, 2020-04-14:
I configured the rootfs_overlay as it'll start a statically compiled Qt project so you can just freshly create your sdcard image to test it. Just 
1. `make ftdev_rpi3_fastboot_defconfig`
2. `make`
then you'll have a test system that runs Qt app automatically.

ricardojlrufino, 2020-04-20:
Hi,
I will try a new full rebuild, this night

Thanks

furkantokac, 2020-05-30:
I checked and it works without any problem. If you have any problem, please don't hesitate to contact.

closed 2020-05-30 by furkantokac