furkantokac/buildroot #4: Problem with enabling USB & Ethernet & Wlan support

mehranmd, 2020-08-05:
Hi.
Congratulations on the job well done. performance on the base image is really good and even ready to use for pure embedded systems. 
I need to modify the build root to support USB & Ethernet & Wlan on rpi3 b+ but unfortunately after trying many Q&A posts on the internet and forums failed to achieve my goal.

I will point out the articles that I used and what I did.

1_http://nightly.buildroot.org/manual.html
on this article, I read it all carefully and mostly focused on the "6.3. init system" part. I understood that I need to enable the
"ifup down scripts' in 'make menuconfig in order to make ifupdown available from the initab.
ofcurse we need to activate other options in this page to have proper access to network functionality but I'm not going to point them out since they are already activated in the default configuration.

2_https://jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Buildroot.html
Another build root but with long loading time (20 sec with class 10 sd card) but fully functional.
i noticed there are few differences in 'linux-menuconfig' command pannel. i assume it's because of the different version of the Linux kernel they used, any way I tried to hard copy the settings manually from this build root to yours but outcome was failure ofcurse. i tried to use the .config which is generated in the linux-(some odd code) instead of the original one and the os which came out of those configs wont even boot now. (didn't had much hope on this method at the start but I had to try)

3_https://git.busybox.net/busybox/tree/examples/inittab
I tried to understand the inittab mechanism in busybox and figured out it's working just fine and there is nothing I could do to improve/fix my situation.

4_https://subscription.packtpub.com/book/hardware_and_creative/9781783289455/1/ch01lvl1sec09/generating-a-basic-root-filesystem-must-know
Really long article on the matter of building root from scratch. it was just hard to relate the article sections to the your build root 
but I can tell there is nothing missing on the matter.

so in **short** I have the drivers there from the 'linux-menuconfig' and proper packages from the 'menuconfig'. but my problem is still standing.
Any share of thoughts and opinions would be extremely welcome and appreciated.

mehranmd, 2020-08-06:
Hi again, today i had some progress and managed to enable USB devices like keyboard and mouse. but the flash disk won't go into the /dev/sdX . I believe I need to enable uDev and euDev in advance to make the flash disk work properly but in the 'menuconfig' the udev and eudev are disabled with the error 
`*** eudev needs eudev /dev management *** `
any idea what is wrong in the process, do I need to enable something else in order to install uDev?

ps: I will post my configurations after I'm done.

furkantokac, 2020-08-10:
Hey! I'm busy in these days so couldn't check your work. Could you solve the issue ?

mehranmd, 2020-08-12:
Hi, Thanks for your consideration.
I managed to enable the network and DHCP with 4sec boot time. but the network won't work as expected. literally no packet goes through.
Today I'm going to work on that issue and then I go after the sound card.
Again I will share my config for every module I enable hopefully it will be useful to someone else.

furkantokac, 2020-08-12:
Nice work! Happy to see your progress.

If your boot process is blocked by a network connection establishment process (DHCP etc.), you can do it after your fastboot required process is run. Userspace initialization is handled by Busybox so you can either replace Busybox with your program (just put your program to /sbin/init) and do the required initializations yourself or you can control Busybox's init processes from /etc/init.d.

mehranmd, 2020-08-13:
True indeed. dhcp is really time consuming. right now I don't use that for setting up the network. just getting things done manually:
```
ifconfig eth0 up
ifconfig eth0 IPADRESS
add route default gw DEFAULTGW
```

mehranmd, 2020-08-13:
Right now im facing a problem building bcm2835 kernel audio module but there is a error stopping me to do so.

`
drivers/char/broadcom/vc_sm/vmcs_sm.o: In function `bcm2835_vcsm_probe':
vmcs_sm.c:(.text+0xf44): undefined reference to `vchiq_add_connected_callback'
drivers/char/broadcom/vc_sm/vmcs_sm.o: In function `vc_sm_connected_init':
vmcs_sm.c:(.text+0x1464): undefined reference to `vchi_initialise'
vmcs_sm.c:(.text+0x1480): undefined reference to `vchi_connect'
drivers/char/broadcom/vc_sm/vc_vchi_sm.o: In function `vc_vchi_sm_videocore_io':
vc_vchi_sm.c:(.text+0xa0): undefined reference to `vchi_service_release'
vc_vchi_sm.c:(.text+0xb8): undefined reference to `vchi_service_use'
vc_vchi_sm.c:(.text+0x114): undefined reference to `vchi_queue_kernel_message'
vc_vchi_sm.c:(.text+0x180): undefined reference to `vchi_msg_peek'
vc_vchi_sm.c:(.text+0x1dc): undefined reference to `vchi_msg_remove'
drivers/char/broadcom/vc_sm/vc_vchi_sm.o: In function `vc_vchi_sm_init':
vc_vchi_sm.c:(.text+0x3e4): undefined reference to `vchi_service_open'
vc_vchi_sm.c:(.text+0x460): undefined reference to `vchi_service_close'
drivers/char/broadcom/vc_sm/vc_vchi_sm.o: In function `vc_vchi_sm_stop':
vc_vchi_sm.c:(.text+0x4bc): undefined reference to `vchi_service_use'
vc_vchi_sm.c:(.text+0x4c4): undefined reference to `vchi_service_close'
Makefile:1012: recipe for target 'vmlinux' failed
make[2]: *** [vmlinux] Error 1
package/pkg-generic.mk:238: recipe for target '/media/mehran/extra2/speedPi/buildroot/output/build/linux-c117a8bccf37bfba323065b566cf999ed4629a4a/.stamp_built' failed
make[1]: *** [/media/mehran/extra2/speedPi/buildroot/output/build/linux-c117a8bccf37bfba323065b566cf999ed4629a4a/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
`
Any idea what is wrong?

mehranmd, 2020-08-15:
I'm facing a problem with sound card functionality which I need help with.
I installed snd_bcm2835 drivers successfully and installed all of the alsa utility packages + gstreamer1 + pulse audio.
Now I use 
`modprobe snd_bcm2835`
to probe the rpi3 b+ sound card and it will list in lsmod afterward but it will not list in /dev/snd which will lead to alsamixer not detecting the sound card at all.
Any idea or suggestions would be welcome. 

reffrence i found for the matter:

[Link1](https://www.raspberrypi.org/forums/viewtopic.php?t=144523)
conclusion: add 'dtparam=audio=on' to /boot/config.txt but it wont help in my case.

[Link2](https://mike632t.wordpress.com/2013/09/09/enable-analogue-audio-output-raspberry-pi/)
conclusion: what is lsmod and probmode command and how you can prob a driver.

mehranmd, 2020-08-17:
I believe my sound problem is coming from kernel not receiving my 'dtparam=audio=on' in boot config. how can I ensure that this parameter is passed?

furkantokac, 2020-08-19:
@mehranmd You can check the beginning of "dmesg" output to see kernel parameters. The line should include something like "root=".

How do you test the sound ? I suggest you to use Alsa and test the sound with an .ogg file and play it by "aplay". Please check [this](https://github.com/furkantokac/buildroot/blob/tk1/br2-external-ftdev/configs/ftdev_tk1_defconfig) to see how I enabled sound for TK1. By this config, sound is usable by QtMultimedia.

mehranmd, 2020-08-19:
Thnx for the share, ill check it out as soon as I can.
Right now I made really good progress with the mix of your [excellent article](https://www.furkantokac.com/rpi3-fast-boot-less-than-2-seconds/)  and the newest BuildRoot (2012.02).
Ethernet is OK.
USB Storage support is OK.
Sound card is OK.
Wlan is NOT OK.
the boot time is 4.4 sec right now which for my case is acceptable.

mehranmd, 2020-08-24:
I just manually copied the /opt folder from base raspbian and rebuilt the raspbian.

changes in the config output:
```
QPA backends:
EGLFS Raspberry Pi ................... yes
```

furkantokac, 2020-08-27:
It should require no Raspian files, actually. I tested it and it works without any issue. There should be another problem.

mehranmd, 2020-09-09:
Hi again.
I'm back to work on the Buildroot again, there is one last problem with my build which is after appending dtb file to the zImage the audio stops working. i compared kernel input commands and there is no difference there and when I load snd_bcm2835  manually the os won't detect the sound card.
looks like overlay of the rpi is stopped working after appending the dtb file to the zImage.
related problem:
[Loading of kernel module with overlay](https://www.raspberrypi.org/forums/viewtopic.php?t=214202)

mehranmd, 2020-09-09:
dmesg with the Buildroot kernel before appending dtb files:
 

>  3.755207] udevd[103]: starting eudev-3.2.9
> [    3.802586] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
> [    3.819042] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
> [    3.842445] hub 1-1.1:1.0: USB hub found
> [    3.853656] hub 1-1.1:1.0: 3 ports detected
> [    3.867214] vchiq: module is from the staging directory, the quality is unknown, you have been warned.
> [    3.888497] vchiq: vchiq_init_state: slot_zero = (ptrval), is_master = 0
> [    3.938989] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
> [    3.957565] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
> [    3.971302] usb 1-1.3: new low-speed USB device number 4 using dwc_otg
> [    3.973272] [vc_sm_connected_init]: start
> [    3.997283] [vc_sm_connected_init]: installed successfully
> [    4.050396] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
> **[    4.070678] bcm2835_audio soc:audio: card created with 8 channels**

dmesg after appending:

> [    3.907642] udevd[103]: starting eudev-3.2.9
> [    3.977511] usb 1-1.3: New USB device found, idVendor=093a, idProduct=2510, bcdDevice= 1.00
> [    3.992137] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [    4.005281] vchiq: module is from the staging directory, the quality is unknown, you have been warned.
> [    4.006094] usb 1-1.3: Product: USB Optical Mouse
> [    4.033152] usb 1-1.3: Manufacturer: PixArt
> [    4.045333] vchiq: vchiq_init_state: slot_zero = (ptrval), is_master = 0
> [    4.061777] input: PixArt USB Optical Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:093A:2510.0001/input/input0
> [    4.088312] hid-generic 0003:093A:2510.0001: input,hidraw0: USB HID v1.11 Mouse [PixArt USB Optical Mouse] on usb-3f980000.usb-1.3/input0
> [    4.114584] usb 1-1.1.2: new low-speed USB device number 5 using dwc_otg
> [    4.134025] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
> [    4.151817] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
> [    4.165180] [vc_sm_connected_init]: start
> [    4.176720] [vc_sm_connected_init]: installed successfully
> [    4.215620] cfg80211: Loading compiled-in X.509 certificates for regulatory database
> [    4.310748] usb 1-1.1.2: New USB device found, idVendor=258a, idProduct=0001, bcdDevice= 1.00
> [    4.326623] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [    4.341704] usb 1-1.1.2: Product: USB KEYBOARD
> [    4.353537] usb 1-1.1.2: Manufacturer: SINO WEALTH
> [    4.372238] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'

from the looks of it kernel is not detecting the sound card after appending the dtp files.
I'm really suspicious with the **overlay** folder in the boot partition, maybe I need to apeend that too but a folder is not the input of the command:
`cat zImage myboard.dtb > my-zImage`

furkantokac, 2020-09-17:
Hey @mehranmd ,

Overlays should not require appending to be processed. There should be related with another problem. I recommend you this;

1. `dmesg` to kernel without appended dtb, which plays the sound
2. `dmesg` to kernel with appended dtb, which doesn't plays the sound
3. From the beginning of `dmesg` logs, check what is different, what different parameters are applied to the sound-available kernel. This should give some clue about the problem.

Waiting to your good new!

furkantokac, 2020-09-20:
I just though that overlays may not be processed by start.elf when we have appended-dtb. You can verify this like that;

1. Run the system with un-appended dtb + overlays. Sound should run without any problem as you said.
2. Run the system with un-appended dtb. Check if you have the sound. If you don't have the sound, find which overlay is responsible for enabling soundcard. Overlays: https://github.com/raspberrypi/firmware/tree/master/boot/overlays
3. Merge sound-overlay to dtb. To do this, simply convert dtbo to dts and copy it to main dts file, after all convert main dts file to dtb file.
4. Append new dtb file to Linux image.

This should work.

mehranmd, 2020-09-21:
Well, I did those tests exactly and the sound works in un-appended version of my kernel image.
I found related problems in the link below but I will test what you suggested as well.
[Overlay load problem](https://www.raspberrypi.org/forums/viewtopic.php?t=134018)

furkantokac, 2020-09-21:
New version of uboot supports overlays but the thing is, Raspberry's boot process is kind of different than a common boot process. In Raspberry, start.elf processes dtb and overlays and do the necessary settlements about the device before booting the kernel. If you append dtb to kernel and remove it in boot partition, you should handle the things manually that start.elf would apply by processing dtb and overlays.

closed 2020-12-09 by furkantokac

McuMirror, 2022-08-27:
> Hi, Thanks for your consideration. I managed to enable the network and DHCP with 4sec boot time. but the network won't work as expected. literally no packet goes through. Today I'm going to work on [...]

hi mehranmd,
Can you share the your buildroot config.? As you promise.