furkantokac/buildroot #5: buildroot don't build

demirkolerme, 2020-09-17:
`/usr/bin/make -j1 O=/home/emre/os/os2/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: uyarı: alt derlemede -jN zorlandı: iş sunucusu kipi kapatılıyor.
>>> glibc glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89 Building
PATH="/home/emre/os/os2/buildroot/output/host/bin:/home/emre/os/os2/buildroot/output/host/sbin:/home/emre/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"  /usr/bin/make   -C /home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build
/usr/bin/make -r PARALLELMFLAGS="" -C /home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89 objdir=`pwd` all
/usr/bin/make  subdir=csu -C csu ..=../ subdir_lib
/usr/bin/make  subdir=iconv -C iconv ..=../ subdir_lib
/usr/bin/make  subdir=locale -C locale ..=../ subdir_lib
/usr/bin/make  subdir=localedata -C localedata ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=iconvdata -C iconvdata ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=assert -C assert ..=../ subdir_lib
/usr/bin/make  subdir=ctype -C ctype ..=../ subdir_lib
/usr/bin/make  subdir=intl -C intl ..=../ subdir_lib
/usr/bin/make  subdir=catgets -C catgets ..=../ subdir_lib
/usr/bin/make  subdir=math -C math ..=../ subdir_lib
/usr/bin/make  subdir=setjmp -C setjmp ..=../ subdir_lib
/usr/bin/make  subdir=signal -C signal ..=../ subdir_lib
/usr/bin/make  subdir=stdlib -C stdlib ..=../ subdir_lib
/usr/bin/make  subdir=stdio-common -C stdio-common ..=../ subdir_lib
/usr/bin/make  subdir=libio -C libio ..=../ subdir_lib
/usr/bin/make  subdir=dlfcn -C dlfcn ..=../ subdir_lib
/usr/bin/make  subdir=malloc -C malloc ..=../ subdir_lib
/usr/bin/make  subdir=string -C string ..=../ subdir_lib
/usr/bin/make  subdir=wcsmbs -C wcsmbs ..=../ subdir_lib
/usr/bin/make  subdir=timezone -C timezone ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=time -C time ..=../ subdir_lib
/usr/bin/make  subdir=dirent -C dirent ..=../ subdir_lib
/usr/bin/make  subdir=grp -C grp ..=../ subdir_lib
/usr/bin/make  subdir=pwd -C pwd ..=../ subdir_lib
/usr/bin/make  subdir=posix -C posix ..=../ subdir_lib
/usr/bin/make  subdir=io -C io ..=../ subdir_lib
/usr/bin/make  subdir=termios -C termios ..=../ subdir_lib
/usr/bin/make  subdir=resource -C resource ..=../ subdir_lib
/usr/bin/make  subdir=misc -C misc ..=../ subdir_lib
/usr/bin/make  subdir=socket -C socket ..=../ subdir_lib
/usr/bin/make  subdir=sysvipc -C sysvipc ..=../ subdir_lib
/usr/bin/make  subdir=gmon -C gmon ..=../ subdir_lib
/usr/bin/make  subdir=gnulib -C gnulib ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=wctype -C wctype ..=../ subdir_lib
/usr/bin/make  subdir=manual -C manual ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=shadow -C shadow ..=../ subdir_lib
/usr/bin/make  subdir=gshadow -C gshadow ..=../ subdir_lib
/usr/bin/make  subdir=po -C po ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=argp -C argp ..=../ subdir_lib
/usr/bin/make  subdir=nptl -C nptl ..=../ subdir_lib
/usr/bin/make  subdir=rt -C rt ..=../ subdir_lib
/usr/bin/make  subdir=conform -C conform ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=debug -C debug ..=../ subdir_lib
/usr/bin/make  subdir=mathvec -C mathvec ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=support -C support ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=crypt -C crypt ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=nptl_db -C nptl_db ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=inet -C inet ..=../ subdir_lib
/usr/bin/make  subdir=resolv -C resolv ..=../ subdir_lib
/usr/bin/make  subdir=nss -C nss ..=../ subdir_lib
/usr/bin/make  subdir=hesiod -C hesiod ..=../ subdir_lib
make[4]: Nothing to be done for 'subdir_lib'.
/usr/bin/make  subdir=sunrpc -C sunrpc ..=../ subdir_lib
/usr/bin/make  subdir=nis -C nis ..=../ subdir_lib
[... 178 lines cut ...]
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.o] Error 1
/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.c:88:26: error: implicit declaration of function 'IXDR_GET_U_iNT'; did you mean 'IXDR_GET_U_INT32'? [-Werror=implicit-function-declaration]
    objp->rq_bhardlimit = IXDR_GET_U_iNT(buf);
                          ^~~~~~~~~~~~~~
                          IXDR_GET_U_INT32
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.o] Error 1
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.o] Error 1
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xnfs_prot.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xnfs_prot.o] Error 1
Makefile:258: recipe for target 'sunrpc/others' failed
make[3]: *** [sunrpc/others] Error 2
Makefile:9: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:238: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/.stamp_built' failed
make[1]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
`

hata bu şekilde

furkantokac, 2020-09-17:
Buildroot için gerekli paketleri kurdunuz mu ?

https://buildroot.org/downloads/manual/manual.html#requirement-mandatory

demirkolerme, 2020-09-18:
gerekli tüm paketler kurulu fakat hata devam ediyor 

`cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.o] Error 1
make[4]: *** Bitmemiş işler için bekliyor....
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.o] Error 1
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.o] Error 1
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.o] Error 1
cc1: all warnings being treated as errors
../o-iterator.mk:9: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xnfs_prot.o' failed
make[4]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xnfs_prot.o] Error 1
Makefile:258: recipe for target 'sunrpc/others' failed
make[3]: *** [sunrpc/others] Error 2
Makefile:9: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:238: recipe for target '/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/.stamp_built' failed
make[1]: *** [/home/emre/os/os2/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
`

furkantokac, 2020-09-20:
Yukarıdaki log hatanın belirtildiği log değil, hatanın belirtildiği log, logun daha öncesinde olmalı.

Bu konuda sizin çalışma yapıp hatanızı çözmeye yönelik daha somut ve spesifik sorular sormanız çözüme ulaşılmasını hızlandıracaktır.

emartisoft, 2020-10-07:
Benzer bir durum ile karşılaştım. Mandatory paketler sistemimde kuruludur. Sanal makine üzerinden derleme yapılırken aldığım hata aşağıdadır. Bu arada çok iyi bir iş çıkarmışsın, tebrikler.

**make ftdev_rpi3_fastboot_defconfig** sonrası **make -j8** komutu ile ekran çıktısı şu şekildedir.

`
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.c: In function 'xdr_rex_start':
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.c:48:4: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
    IXDR_PUT_U_iNT(buf, objp->rst_port0);
    ^~~~~~~~~~~~~~
    IXDR_PUT_U_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c: In function 'xdr_statstime':
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:66:5: error: implicit declaration of function 'IXDR_PUT_iNT'; did you mean 'IXDR_PUT_LONG'? [-Werror=implicit-function-declaration]
     IXDR_PUT_iNT(buf, *genp++);
     ^~~~~~~~~~~~
     IXDR_PUT_LONG
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.c:78:22: error: implicit declaration of function 'IXDR_GET_U_iNT'; did you mean 'IXDR_GET_U_INT32'? [-Werror=implicit-function-declaration]
    objp->rst_port0 = IXDR_GET_U_iNT(buf);
                      ^~~~~~~~~~~~~~
                      IXDR_GET_U_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:77:3: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
   IXDR_PUT_U_iNT(buf, objp->v_pgpgin);
   ^~~~~~~~~~~~~~
   IXDR_PUT_U_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c: In function 'xdr_klm_holder':
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:55:4: error: implicit declaration of function 'IXDR_PUT_iNT'; did you mean 'IXDR_PUT_LONG'? [-Werror=implicit-function-declaration]
    IXDR_PUT_iNT(buf, objp->svid);
    ^~~~~~~~~~~~
    IXDR_PUT_LONG
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:142:15: error: implicit declaration of function 'IXDR_GET_iNT'; did you mean 'IXDR_GET_INT32'? [-Werror=implicit-function-declaration]
     *genp++ = IXDR_GET_iNT(buf);
               ^~~~~~~~~~~~
               IXDR_GET_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:56:4: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
    IXDR_PUT_U_iNT(buf, objp->l_offset);
    ^~~~~~~~~~~~~~
    IXDR_PUT_U_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:73:17: error: implicit declaration of function 'IXDR_GET_iNT'; did you mean 'IXDR_GET_INT32'? [-Werror=implicit-function-declaration]
    objp->svid = IXDR_GET_iNT(buf);
                 ^~~~~~~~~~~~
                 IXDR_GET_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:153:20: error: implicit declaration of function 'IXDR_GET_U_iNT'; did you mean 'IXDR_GET_U_INT32'? [-Werror=implicit-function-declaration]
   objp->v_pgpgin = IXDR_GET_U_iNT(buf);
                    ^~~~~~~~~~~~~~
                    IXDR_GET_U_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:74:21: error: implicit declaration of function 'IXDR_GET_U_iNT'; did you mean 'IXDR_GET_U_INT32'? [-Werror=implicit-function-declaration]
    objp->l_offset = IXDR_GET_U_iNT(buf);
                     ^~~~~~~~~~~~~~
                     IXDR_GET_U_INT32
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.c: In function 'xdr_rquota':
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.c:50:4: error: implicit declaration of function 'IXDR_PUT_iNT'; did you mean 'IXDR_PUT_LONG'? [-Werror=implicit-function-declaration]
    IXDR_PUT_iNT(buf, objp->rq_bsize);
    ^~~~~~~~~~~~
    IXDR_PUT_LONG
cc1: all warnings being treated as errors
/home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrquota.c:52:4: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
    IXDR_PUT_U_iNT(buf, objp->rq_bhardlimit);
    ^~~~~~~~~~~~~~
    IXDR_PUT_U_INT32
make[4]: *** [../o-iterator.mk:9: /home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.o] Error 1
make[4]: *** Bitmemiş işler için bekliyor....
[... cut ...]
cc1: all warnings being treated as errors
make[4]: *** [../o-iterator.mk:9: /home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xnfs_prot.o] Error 1
make[3]: *** [Makefile:258: sunrpc/others] Error 2
make[2]: *** [Makefile:9: all] Error 2
make[1]: *** [package/pkg-generic.mk:241: /home/emarti/Rpi3/buildroot/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/.stamp_built] Error 2
make: *** [Makefile:84: _all] Error 2

`

furkantokac, 2020-10-07:
Gönderdiğiniz hata derleyici sürümü ile alakalı olabilir. Kod bir derleyici sürümünün baz aldığı standarta göre yazılıyor, dolayısıyla aynı kod farklı sürüm derleyici ile derlenmeye çalışılınca hata çıkabiliyor.

Ubuntu 20.04 (gcc 9.3.0) tabanlı bir dağıtımda test ediyorum, sıkıntısız çalışıyor. Zaten bu repoyu uzun süredir aktif kullanmaktayım. Sanal makinede hiç denemedim. `make -j8` yerine `make` şeklinde tek thread ile derlemeyi deneyebilirsiniz.

Bunun yanında derleme ortamı ile ilgili daha detaylı bilgiler ve tam log dökümü paylaşabilirseniz daha iyi olacaktır. Hatalar logların son kısmından ziyade daha öncesinde yazıyor.

closed 2020-10-20 by furkantokac