furkantokac/buildroot #7: Raspberry Pi 3 fast boot img ready - Run Python app piSerial ttyS0 fast

bass99, 2020-10-17:
Hi Furkan,
I am a big fan of raspberry pi and Python I have developed many scripts in Python Serial ttyX to extract data registers from controllers.

Now for a critical application I would need my Python serial application active in a few seconds, unfortunately even with raspberry pi 4 it takes an average of 5-7 seconds to initialize my script (I use the systemd linux daemon to activate my scripts).

I have to read from the serial and give immediate printing instructions on an ESC-POS label printer device, in a maximum of 2 seconds from power on my program must already be running on the raspberry.

Now I downloaded your fastboot raspberry image in 2 seconds, I wanted to ask you if you can use my scripts already, Python 3.7 is installed, can my python 3.7 script be run?

They tell me that ESP type microcontrollers are much faster to boot but I would like to stay on a fast raspberry linux board.

thanks for now
Rocco

furkantokac, 2020-10-18:
Hey Rocco,

If there is no graphical application to run, the current work hits 1.25sec from power-up to Linux userspace with serial connection enabled, by using start_cd.elf.

In your use-case, Python interpreter initialization and your Python script initialization will be run after Linux is ready. To run Python script, we can have different strategies. The first thing comes my mind is as the following;

I would statically cross compile CPython for the target device and try to embed my script in it. Maybe even you can directly convert your Python script to executable file, this could be easier but less flexible. After that I would have a single file of statically compiled CPython and my script is embedded in it.

1. (0.75sec) Hardware boot (start_cd.elf is used)
2. (0.25sec) Linux pre-boot
3. (0.25sec) Linux boot (till here, you can use ftdev_rpi3)
4. Statically compiled file (just put it to /sbin/init). If you need to initialize anything, you need to do it manually in statically compiled file at this point. I don't really know how much time this part will take.

It's fun to work on this. Good job. Don't hesitate to contact!

Regards,
Furkan

bass99, 2020-10-18:
Hello Furkan,
my Python MODBUS script is already be executable, command
sudo python3 myScript.py

The script is now automatically executed at startup by the powerful Systemd LINUX daemon!

Now I'm sorry I didn't quite understand what well what you mean with Cpython (I know python not cpython), embed a script in it what does it mean?

What should I incorporate, remember that it is the raspberry machine (MASTER) that simply executes the PYTHON script, not the MODBUS (SLAVE) card which is instead interrogated.

I have now written your rasbian image on the microsd, could you kindly describe me the steps to start the raspberry machine from ssh terminal.
I would like to start testing it on the raspberry machine via ssh terminal:
1) can i start raspberry from ssh? I remember that I program the raspberry from the ssh terminal after configuring the wlan0 of the linux machine (etc / network / interfaces)
2) where do I configure wlan0 for the first boot ssh access via terminal?
3) Is Python 3.7 already installed in your raspberry image? if you can't install with pip?

If I can get the Python Serial application to go fast start with your special rom I would like to stay so long on the raspberry machine, esp8266 I don't know it.

I am attaching the minilmodbus python script (powerful Python libraries) which reads and writes the registers on the modbus slave board.

thanks for now
Rocco

Script Python MODBUS:

import minimalmodbus
import time
import threading
import subprocess
import logging
import serial
import socket
import fcntl
import struct

instrument = minimalmodbus.Instrument("/dev/ttyS0", 1) # port  slave address , 1=ID SLAVE
instrument.serial.baudrate = 4800   # Baud
instrument.serial.bytesize = 8
##instrument.serial.parity   = PARITY_NONE
instrument.serial.stopbits = 1
instrument.serial.timeout  = 0.50   # seconds
instrument.debug = False #Da testare "True" in caso di errore IOError Communication

while True:
    try:

        ## Read temperatura registro 102 ----> signed=True
        t1 = instrument.read_register(102,0,3,signed=True) 
        print "t1=",str(t1)
        time.sleep(0.01)

        ## Read temperatura registro 103 ----> signed=True
        t2 = instrument.read_register(103,0,3,signed=True)
        print "t2=",str(t2)
        time.sleep(0.01)

        ## Read version software ---> intero ----> signed=False
        versionSoft= instrument.read_register(136,0,3,signed=False) 
        print "versionSoft=",str(versionSoft)
        time.sleep(0.01)

        ## Read ai[A/ora] istantanei con Decimals=1 --->  signed=False
        ai = instrument.read_register(108,1,3,signed=False) 
        print "ai=",str(ai)
        time.sleep(0.01)

        ## Read ah[Kwh]=32bit=16bit_hight+16bit_low ---> signed=False
        ah = instrument.read_long(110,3,signed=False)
        print "ah=",str(ah)
        time.sleep(0.01)

        ## Read h1 long 32 bit
        h1= instrument.read_long(128,3,signed=False)
        print "h1=",str(h1)
        time.sleep(0.01)

        ## Read h2 long 32 bit=
        h2 = instrument.read_long(130,3,signed=False)
        print "h2=" ,str(h2)
        time.sleep(0.01)

        ## Read h3=2 registri da 16 bit
        h3 = instrument.read_long(132,3,signed=False)
        print "h3=",str(h3)
        time.sleep(0.01)

        ## Read h4=2 registri da 16 bit=32bit=16bit_hight+16bit_low
        h4 = instrument.read_long(134,3,signed=False)
        print "h4=" ,str(h4)
        time.sleep(0.01)

        ## Read t1_max
        t1_max = instrument.read_register(104,0,3,signed=True)
        print "t1_max=",str(t1_max)
        time.sleep(0.01)

        ## Read t1_min
        t1_min = instrument.read_register(105,0,3,signed=True)
        print "t1_min=",str(t1_min)
        time.sleep(0.01)

        ##Read t2_max
        t2_max = instrument.read_register(106,0,3,signed=True)
        print "t2_max=",str(t2_max)
        time.sleep(0.01)

        ## Read stato dig. 1 (Function INVERSIONE stato digitale)
        i1 = int(not instrument.read_register(127,0,3,signed=False))
        time.sleep(0.01)
        print "i1=",i1

        ## Read stato digitale 2
        i2 = int(not instrument.read_register(126,0,3,signed=False))
        time.sleep(0.01)
        print "i2=",i2

        ## Read stato digitale 3
        i3 = int(not instrument.read_register(125,0,3,signed=False))
        time.sleep(0.01)
        print "i3=",i3

        ## Read stato digitale 4
        i4 = int(not instrument.read_register(124,0,3,signed=False))
        time.sleep(0.01)
        print "i4=",i4

        ## Read stato digitale 5
        i5 = int(not instrument.read_register(123,0,3,signed=False))
        time.sleep(0.01)
        print "i5=",i5

        ## **************WRITE_REGISTER_MODBUS **********************
        ## write registro 16 bit e lettura
        # t1max=30
        # instrument.write_register(104, t1max,0,16,signed=False) 
        # t1max_write = instrument.read_register(104, 0,3,signed=True) 
        # print "t1max_write=",str(t1max_write)
        # time.sleep(0.15)

        ## write registro 32 bit e lettura
        # impulsi2=450000
        # instrument.write_long(114, impulsi2,signed=False) 
        # impulsi2_write = instrument.read_long(114,3,signed=False) 
        # print "impulsi2_write",str(impulsi2_write)
        # time.sleep(0.15)

        ##WRITE ser[long]=32bit=16bit_hight+16bit_low
        #ser_w=120
        #instrument.write_long(100,ser_w,signed=False)
        #ser_w = instrument.read_long(100,3,signed=False)
        #print "ser_w=",str(ser_w)
        #time.sleep(0.15)
        

        print '*' * 100 + "\n"

    ## Capture Exception IOError
    except IOError:
        print IOError

    ## Capture Exception ValueError
    except  ValueError:
        print ValueError

``

furkantokac, 2020-12-04:
> The script is now automatically executed at startup by the powerful Systemd LINUX daemon!

In fastboot, you need control and minimalism so big projects like systemd is bad for fastboot. There are fits-better-to-fastboot-application init systems. More of that, I recommend you to cancel init system at all, if you can.

------

> Now I'm sorry I didn't quite understand what well what you mean with Cpython (I know python not cpython), embed a script in it what does it mean?

Its the source code of the Python interpreter. Check it online for details.

------

> I have now written your rasbian image on the microsd, could you kindly describe me the steps to start the raspberry machine from ssh terminal.

If you don't have to use ssh, use serial connection: https://www.instructables.com/How-to-Access-Headless-Raspberry-Pi-Using-USB-to-T/

------

> 1. can i start raspberry from ssh? I remember that I program the raspberry from the ssh terminal after configuring the wlan0 of the linux machine (etc / network / interfaces)

There are several strategies for that. You can search about this by the following keywords: raspberry network boot

------

> where do I configure wlan0 for the first boot ssh access via terminal?

This is not really related with fastboot. It depends on your userspace network manager. For example, I use the following settings for a system with systemd and connman

Copy the following 2 files into the /var/lib/connman;

`connman-settings.config`
```
[global]
OfflineMode=false

[Wired]
Enable=true
Tethering=false

[WiFi]
Enable=true
Tethering=false
```

`connection-settings.config`
```
# Your wifi info
[service_ftwifi]
Type = wifi
Name = yourwifiname
Passphrase = yourwifipassword
IPv4 = dhcp
```

I wrote a script that checks usb wifi and reset it if the connection is broken;

`ftwifi`
```python
#!/usr/bin/python3

import time
import threading
import utils as u

class FtWifi(object):
        def __init__(self):
                self.adapter_port = "1-1.3"

                self.thread = threading.Thread(target=self.loop)
                self.__is_running = False

        def plug_adapter(self):
                u.run_cmd("echo " + self.adapter_port + " > /sys/bus/usb/drivers/usb/bind")

        def unplug_adapter(self):
                u.run_cmd("echo " + self.adapter_port + " > /sys/bus/usb/drivers/usb/unbind")

        def replug_adaptor(self):
                self.unplug_adapter()
                time.sleep(3)
                self.plug_adapter()
                time.sleep(3)

        ###
        ### Run functions
        ###
        def loop(self, is_threaded=True):
                self.__is_running = not is_threaded
                sleep_time = 10
                while self.__is_running:
                        self.run()
                        time.sleep(sleep_time)

        # Return True if online, False if offline.
        def run(self):
                is_connected = u.is_connected("wlan0")
                if not is_connected: is_connected = u.is_connected("eth0")

                if not is_connected:
                        u.log("Disconnected. Repluging the adapter.")
                        self.replug_adaptor()

                return is_connected

        def start(self):
                if self.__is_running: return

                self.__is_running = True
                self.thread.start()

        def stop(self):
                self.__is_running = False

if __name__ == "__main__":
        app = FtWifi()
        app.loop(is_threaded=False)

```

I have a systemd service that runs the script above;

`ftwifi.service`
```
[Unit]
Description=FT USB wifi handler.

[Service]
Type=oneshot
RemainAfterExit=yes                             
ExecStart=/bin/ftwifi

[Install]            
WantedBy=basic.target
```

This was working on iMX6. It was okay for my usecase, you should create your own solution that fits to your own usecase.

----

> Is Python 3.7 already installed in your raspberry image? if you can't install with pip?

No but you can install whatever you want by your own. This is a base image.

----

To sum up, what you're doing requires a plan and time. Please also read my previous message by checking the terms online. Hope that you can handle it! You can update your status here to let us know the progress.

closed 2020-12-04 by furkantokac

bass99, 2020-12-05:
Hi Furkan,
I don't understand well, systemd can be used on your fast rom?
Is present?

I would simply like to understand the quick steps to use your quick rom on
the raspberry and run the Python script with systemd or other system.

if you can better specify the steps for your rom run
thanks for now
R