furkantokac/buildroot #8: Additional optimizations

IronOxidizer, 2020-10-17:
I'm attempting a similar project for the Raspberry Pi Zero. Here are some additional optimizations I have found:

- Use LZ4 kernel compression for faster boot times
    - https://events.static.linuxfound.org/sites/events/files/lcjpcojp13_klee.pdf
- Use `fastboot` mount option for F2FS to improve boot times, also consider `sync`
    - https://www.kernel.org/doc/Documentation/filesystems/f2fs.txt
    - https://linux-f2fs-devel.narkive.com/cND2je9J/f2fs-mount-o-sync-sync-and-cut-off-power
    - I noticed that the very first boot after a clean flash with F2FS is usually 10% faster than the following boots. These flags should fix this issue resulting in a speed up.
- Statically compile with `musl libc` to produce smaller binaries that load faster
    - https://www.etalabs.net/compare_libcs.html

furkantokac, 2020-10-18:
Thanks for your contribution!

I don't have experience with RPI Zero but it should be similar to RPI3 since they use same series of SoCs. For RPI3, the main bottleneck is the hardware. It takes 1.5sec to boot with a graphic support, 1.0sec to without graphic support. There are some new works about open-source firmware for the RPI3, I'm following them. If we can solve this problem, there is a big potential even faster boot time.

IronOxidizer, 2020-10-19:
Mind pointing me in the direction of that open-source RPi firmware?

I've been experimenting with the cut down versions of the firmware (`start_cd.elf` and `fixup_cd.elf`). I've managed to save 0.3s with it but It's not practical for all uses as it removes a large amount of GPU functionality.

You can check it out here: https://github.com/IronOxidizer/instant-pi

furkantokac, 2020-12-04:
> Mind pointing me in the direction of that open-source RPi firmware?

This is the main issue. If I focus on more progress, I work on in. That would be very cool!

Image compression you mention does not really make big impact on the current image.

I checked your repo, its cool. Good job 👍🏻

closed 2020-12-04 by furkantokac