furkantokac/buildroot #9: Fastboot and WiringPI

kamenmi1, 2020-10-18:
I'm working on the project to make a raspberry super fast and be able to run a application created in cmake. I'm using a raspberry compute modul, display and analog port to communicate with laptop. Communication is done via pppd to reduce amount of services running. The application is mainly using WiringPI library and it's created in Cmake. The hardware is using those sockets to run and show data properly UART0 (for communication with laptop) , UART1, SPI0, SPI2, I2C0 and I2C. Those are all necesary libraries (I2C_library, pthread, wiringPi, wiringPiDev, crypt, rt). The goal is able to run this software and show data to display in 3-5 sec after pressing the power button. 
I'm stuck to make it work for last seven months, so any help or idea will help me a lot. Thanks so much

furkantokac, 2020-10-18:
Nice project.

You can boot even fully-functional distro in 5sec. What you need to do is as the following;

1. Download and compile a RPI distro (ftdev_rpi3_fastboot_defconfig)
2. Statically cross-compile your application by using compiled distro. You should have all the external library sources (wiringPi etc) externally in your project since when you statically compile your application, required sources will be included in the compiled file. At the end you should have a single binary at this point. You may need to solve problems here, you'll need to spend some time at this point. For example, https://github.com/furkantokac/buildroot/blob/ftdev/build-rpi3-qt.sh this script statically cross-compile Qt. 
3. Put your binary file into RPI and you should have less than 2secs of boot time. To speed up more, Just change start.elf in the boot partition to start_cd.elf and you'll have less than 1.5sec of boot time. Crazy enough ? :)

I recommend you that before all, for the 2. step, just use a "Hello World" project without any external libraries, statically cross-compile it and understand the concepts. After that, try to replace "Hello World" project with your project. It is not that hard, you can do it! I'm here till you succeed.

kamenmi1, 2020-10-19:
Hello, 
first of all thank you so much for all of that. I will work on it hard and keep you posted with updates.
I would have just few following questions. 
First one would be about the static cross-compile. Should I use buildroot to compile the application or should I do it outside of the buildroot and then just put the application inside the distro? 
Second one would be about the ftdev_rpi3_fastboot_defconfig. Do I have to modify somehow the buildroot configuration or is it already good enough, so I can simply compile it?
Those would be my two questions at the moment. If I could get to the 1.5 sec, that would be simply mind-blowing. 
Thank you again and sending you greetings.

furkantokac, 2020-10-20:
> First one would be about the static cross-compile. Should I use buildroot to compile the application or should I do it outside of the buildroot and then just put the application inside the distro?

When you create a distro by Buildroot, it gives you all the required tools and files like cross-compiler, rootfs etc. so it makes your life easier, so I recommend you to use it if you don't have a valid excuse.

> Second one would be about the ftdev_rpi3_fastboot_defconfig. Do I have to modify somehow the buildroot configuration or is it already good enough, so I can simply compile it?

It should be enough. This is all about the external libraries and their dependencies (especially drivers). I run Qt application on it without any problem.

> Those would be my two questions at the moment. If I could get to the 1.5 sec, that would be simply mind-blowing.
Thank you again and sending you greetings.

1.5sec is quite possible. What you need to is as the following at first;

1. Write a simple "Hello world" code in C
2. git clone git@github.com:furkantokac/buildroot.git
3. cd buildroot
4. make ftdev_rpi3_fastboot_defconfig
5. make -j8
6. Format your RPI with the new image created
7. Cross-compile your code by cross-compilers from output/host/usr/bin/arm-buildroot-linux-gnueabihf-*
8. Put your binary file to Raspberry. To auto-run it, create a simple script that runs your binary in the /etc/inid./ directory

These are the simple steps. Try to understand the steps so that you can freely do what you want.

Good luck!

kamenmi1, 2020-10-20:
Hello, 
I would like thank you for providing me with such a nice and understandable set of things what I should do to make it run. I very appreciate it ❤️ 

1) I was able to build the ftdev_rpi3_fastboot_defconfig image and place it to my raspberry. 
2) I have created a basic "Hello World" script in C and I tried to cross-complied it. "Tried" is quite important, because even though I think I did it correctly when I place the binary file into raspberry /root folder (same place, where is QmlTestApplication) and write a simple auto-run bash script in /etc/init.d folder - It still does only show four raspberries when I boot the raspberry and connect it to monitor via HDMI. 
3) I have used several cross compilers, but It didn't bring me any success 
for example: 
arm-buildroot-linux-gnueabihf-cc
arm-buildroot-linux-gnueabihf-gcc
arm-buildroot-linux-gnueabihf-gcc-7.4.0
arm-buildroot-linux-gnueabihf-gcc-ar

4)When I run a file command on my cross-compiled binary file I got this output:

helloworld: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 4.14.0, not stripped

5) Despite using different cross-compilers, It still didn't show Hello World after running it in raspberry. I have tested my script by running your QmlTestApplication with it and your application with arrows does show up at the screen after booting raspberry, so I believe there must be some my mistake with cross-compiling. 

6) Sadly, I don't have any logs - or I couldn't find them, so I guess it has been disabled. 

7) For C code, I have used those two pages:
https://www.programiz.com/c-programming/examples/print-sentence
https://www.96boards.org/documentation/guides/crosscompile/commandline.html

I hope that I have explained well. Thank you so much for any feedback :)

furkantokac, 2020-12-20:
I recommend you that, on the development stage, disable your init.d script and connect to console to test your binaries. To connect to console, you can use UART with a small configuration without changing any other thing. I have explained it here: https://github.com/furkantokac/buildroot/issues/14

Most probobly your binary run without any issue, you just don't see the output.

This will help you to make progress. I wait to hear from you.

ucondua, 2021-01-07:
make[3]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase/src/network»
make[3]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase/src/gui»
make[2]: *** [Makefile:569: sub-gui-install_subtargets] Ошибка 2
make[2]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase/src»
make[1]: *** [Makefile:63: sub-src-install_subtargets] Ошибка 2
make[1]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase»
make: *** [Makefile:100: module-qtbase-install_subtargets] Ошибка 2
 Help me please

closed 2021-01-07 by furkantokac