furkantokac/buildroot #10: How add can0 in this fastboot os

Vd2617, 2020-10-19:
We try to work with can in our qt project,how we can add support can0 interface?(hard spi can module for raspbian)

Vd2617, 2020-10-20:
I use socketcan plugin qt,he work with linux socketcan,and mcp2551,how I can add this in your fast boot system?Thanks

furkantokac, 2020-10-20:
This should work;

1. make ftdev_rpi3_fastboot_defconfig
2. make xconfig
3. Enable the following settings, save and exit
```
- BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_EUDEV
- BR2_PACKAGE_IPROUTE2
```
4. make linux-xconfig
5. Enable the following settings, save and exit
```
- NET
- CAN
- CAN_DEV
- SPI
- CAN_MCP251X
```
6. make -j8
7. ./build-rpi3-qt.sh
8. Compile your application and run it in RPI

Vd2617, 2020-10-20:
Thanks,I can try do it

Vd2617, 2020-10-20:
I do it ,but I get errors "error initialization udev control docket
Adress family not support by protocol....
What I can do ?Thank

Vd2617, 2020-10-20:
Can are you help me ,please ?

Vd2617, 2020-10-21:
I see spi doesn't work,in  dev /dev/spi0.0 not found

Vd2617, 2020-10-22:
I add in config.txt 
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25
dtoverlay=spi0-hw-cs

But "lsmod " is empty
I add
modprobe mcp251x
modprobe spi-bcm2835
Then "lsmod" see 
                  can_dev
                  spi_bcm2835
When i add overlay in config text,modules kernel not add automaticaly.
I not see spi0.0 in /dev ,spi doesn't work . What I can do ,to work with spi?

Vd2617, 2020-10-22:
And when I write .img to sd card I not see folder overlay,

Vd2617, 2020-10-25:
I am rewrite .dtb file and add spi and mcp2515,all work,but now can0 devices not found?

Vd2617, 2020-10-28:
Ifconfig not work,socket: Address not support by protocol, ip link up not found can0.....But ip link show  return lo and can0...What are you disable in net support,can you help me ?

Vd2617, 2020-10-28:
I add mcp2515 to dts,and convert to dtb,and add to zimage...in buildroot add ipv6,and some delay to up can0,and all worked:

closed 2020-10-28 by Vd2617

furkantokac, 2020-10-28:
Good job! Can you share the dts or dtb file you used ? Maybe it may be useful to someone.

Vd2617, 2020-11-25:
[bcm2710-rpi-3-b.zip](https://github.com/furkantokac/buildroot/files/5596523/bcm2710-rpi-3-b.zip)

Vd2617, 2020-11-25:
This my DTB with support can interface for spi module to raspberry,i am edit your script and delete your dtb,and add my dtb to kernel

DanielVolt, 2021-07-14:
how do i add your dtb file?

furkantokac, 2021-07-14:
> how do i add your dtb file?

You mean edit or use ?

DanielVolt, 2021-07-14:
> > how do i add your dtb file?
> 
> You mean edit or use ?

yes, sorry, i meant use the dtb file. I just compiled buildroot with the enabled settings you provided, but no ideas how to use the dtb file provided here

Vd2617, 2021-07-15:
Add to buildroot/board/ftdev/rpi3/post-image.sh , path to your dtb , 

...
# Find where is the DTB
if [ -f ${BINARIES_DIR}/bcm2710-rpi-3-b.dtb ]; then
    DTB_FILE="${BINARIES_DIR}/bcm2710-rpi-3-b.dtb" // your path to dtb file
else
    DTB_FILE="${BINARIES_DIR}/rpi-firmware/bcm2710-rpi-3-b.dtb"
fi
...

Vd2617, 2021-07-15:
Or insert my dtb file instead of a "$ {BINARIES_DIR} /bcm2710-rpi-3-b.dtb"

DanielVolt, 2021-09-13:
i have added your DTB file, what is needed more? I still cannot get can0 to function @Vd2617

Vd2617, 2021-09-13:
I don't know much English( If you added a device tree file to the kernel and enabled the kernel module for mcp2515 (modprobe mcp251x
modprobe spi-bcm2835) everything should work. I haven't worked on my project for a long time, it works properly, but in a couple of weeks I will write a new application using can bus, then I can help

DanielVolt, 2021-09-15:
Hey thanks, its okay for you to write in russian i can translate. Quick question: How do you bring up the can0? Do you add in your qt project sudo /sbin/ip link set can0 up type can bitrate 500000?

Vd2617, 2021-09-16:
Я can0 , поднимаю в скрипте перед запуском приложения , скорость использую 250 , так данные на этой скорости приходят с оборудования

DanielVolt, 2021-09-16:
Thank you! Thats is awesome. Can you please share your script?

Vd2617, 2021-09-17:
Исходники и файлы по проекту находятся на накопителе , который не со мною сейчас .У меня остались тестовые образы на sd card , я завтра вам сброшу копию с образа (не помню рабочая ли она )  но можете попробовать ... Если не будет работать , тогда смогу помочь позже

DanielVolt, 2021-09-17:
Awesome, thanks!! Looking forward to the picture. Would really help me out

Vd2617, 2021-09-18:
Send me your email

DanielVolt, 2021-09-18:
My email is danielvolt@outlook.com

keyhun, 2023-01-10:
Hello @Vd2617 
Sorry but I'm still not clear how to apply your DTB
for example if I place your DTB file in /pi/download then am I supposed to change "${BINARIES_DIR}/bcm2710-rpi-3-b.dtb" to "/pi/download/ bcm2710-rpi-3-b.dtb" like this?
I followed all your trace but I still dont see spi0.0 in /sys/bus/spi/devices

What I did are below
At the buildroot terminal "make raspberrypi3_defconfig" then enabled those below via "make xconfig" and "make linux-xconfig"

BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_EUDEV
BR2_PACKAGE_IPROUTE2
NET
CAN
CAN_DEV
SPI
CAN_MCP251X

After that I flashed sdcard.img to an SD card then I add those two lines below to config.txt 
dtparam=spi=on 
dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25
and then 
modprobe mcp251x
modprobe spi-bcm2835
I expected to see spi0.0 but I don't see any device in /sys/bus/spi/devices

Thank you
BR

jasongaunt, 2024-05-01:
For anyone else following in these footsteps, this is how I got this working (all done via terminal, not with X windows);

1. If you are starting with a "used" output, run `make clean`
2. Edit `configs/ftdev_rpi3_fastboot_defconfig` and add 2 new lines:
```ini
BR2_PACKAGE_IPROUTE2=y
BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_EUDEV=y
```
3. Run `make ftdev_rpi3_fastboot_defconfig`
4. Run `make linux-menuconfig` and then make the following changes:
```sh
Enable Networking Support [*]
Enable Networking Support ---> CAN bus system support <M>
Enable Networking Support ---> CAN bus system support  ---> CAN Device Drivers ---> CAN SPI interfaces ---> Microchip MCP251x SPI CAN Controllers <M>
```
5. Add the following lines to `board/ftdev/rpi3/custom_files/config.txt` (changing `oscillator` and `interrupt` values to match your particular MCP251x boards crystal and GPIO pin you connected INT to):
```ini
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=23
dtoverlay=spi-bcm2835-overlay
```
6. Download [the patched DTB file above](https://github.com/furkantokac/buildroot/issues/10#issuecomment-733665807), extract it and place the DTB file into `board/ftdev/rpi3/custom_files/`
7. Edit `board/ftdev/rpi3/post-image.sh` and after line 32 (`# Append device tree to img`) but before `cat ...` add this line:
```sh
DTB_FILE="${BOARD_DIR}/custom_files/bcm2710-rpi-3-b.dtb"
```
... which should then look like this:
```sh
# Append device tree to img
DTB_FILE="${BOARD_DIR}/custom_files/bcm2710-rpi-3-b.dtb"
cat "${BINARIES_DIR}/zImage" "${DTB_FILE}" > "${CUSTOM_OUTPUT_DIR}/zImagen"
```
8. Go back to the main buildroot folder and run `make -j$(nproc)` to build it as fast as your system permits

This *mostly* works - I'm unable to run `ip link set up can0 type can bitrate 1000000` but I'll update this post as soon as I find the answer.

Ps. Gotcha for anyone trying to do 1 Mbit CAN - your oscillator (crystal) must be at least 16 MHz, most come with 8 MHz (their instructions say to add `oscillator=8000000` and these can only do up to a maximum of 500 Kbit