furkantokac/buildroot #13: Raspberry pi 3 ftdev confiği derlenmiyor

PozitronikTech, 2020-11-30:
Furkan selam.

Aşağıda ki işletim sistemi yüklü
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

Senin githubtaki buildroot u indirip **make ftdev_rpi3_fastboot_defconfig** ile config seçimini yapıyorum. Bir kez başarılı derleme görmek için config üzerinde hiçbir değişiklik yapmadan. Derleme işlemini make ile gerçekleştiriyorum. make -j8 8çekirdekli işlemci olması durumunda geçerli olacağını öğrendim. Kullandığım makine 4 çekirdekli lakin sürek kısıtı gözetmeksizin normal bir derleme için make komutunu kullandım. Sonuç itibari ile aşağıda ki hata ile işlem yarım kalmaktadır.
`
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c: In function 'xdr_statstime':
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:66:5: error: implicit declaration of function 'IXDR_PUT_iNT'; did you mean 'IXDR_PUT_LONG'? [-Werror=implicit-function-declaration]
     IXDR_PUT_iNT(buf, *genp++);
     ^~~~~~~~~~~~
     IXDR_PUT_LONG
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:77:3: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
   IXDR_PUT_U_iNT(buf, objp->v_pgpgin);
   ^~~~~~~~~~~~~~
   IXDR_PUT_U_INT32
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:142:15: error: implicit declaration of function 'IXDR_GET_iNT'; did you mean 'IXDR_GET_INT32'? [-Werror=implicit-function-declaration]
     *genp++ = IXDR_GET_iNT(buf);
               ^~~~~~~~~~~~
               IXDR_GET_INT32
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrstat.c:153:20: error: implicit declaration of function 'IXDR_GET_U_iNT'; did you mean 'IXDR_GET_U_INT32'? [-Werror=implicit-function-declaration]
   objp->v_pgpgin = IXDR_GET_U_iNT(buf);
                    ^~~~~~~~~~~~~~
                    IXDR_GET_U_INT32
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c: In function 'xdr_klm_holder':
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:55:4: error: implicit declaration of function 'IXDR_PUT_iNT'; did you mean 'IXDR_PUT_LONG'? [-Werror=implicit-function-declaration]
    IXDR_PUT_iNT(buf, objp->svid);
    ^~~~~~~~~~~~
    IXDR_PUT_LONG
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:56:4: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
    IXDR_PUT_U_iNT(buf, objp->l_offset);
    ^~~~~~~~~~~~~~
    IXDR_PUT_U_INT32
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.c: In function 'xdr_rex_start':
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrex.c:48:4: error: implicit declaration of function 'IXDR_PUT_U_iNT'; did you mean 'IXDR_PUT_U_INT32'? [-Werror=implicit-function-declaration]
    IXDR_PUT_U_iNT(buf, objp->rst_port0);
    ^~~~~~~~~~~~~~
    IXDR_PUT_U_INT32
/home/mustafa/İndirilenler/buildroot-ftdev/output/host/bin/arm-buildroot-linux-gnueabihf-gcc /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrusers.c -c -std=gnu11 -fgnu89-inline  -O2 -Wall -Werror -Wundef -Wwrite-strings -fmerge-all-constants -fno-stack-protector -frounding-math -Wstrict-prototypes -Wold-style-definition -fmath-errno    -Wno-unused -fPIC        -I../include -I/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc  -I/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build  -I../sysdeps/unix/sysv/linux/arm  -I../sysdeps/arm/nptl  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix/arm  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/arm/include -I../sysdeps/arm  -I../sysdeps/wordsize-32  -I../sysdeps/ieee754/flt-32  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I. -nostdinc -isystem /home/mustafa/İndirilenler/buildroot-ftdev/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/7.4.0/include -isystem /home/mustafa/İndirilenler/buildroot-ftdev/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/7.4.0/include-fixed -isystem /home/mustafa/İndirilenler/buildroot-ftdev/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include  -D_LIBC_REENTRANT -include /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/libc-modules.h -DMODULE_NAME=librpcsvc -include ../include/libc-symbols.h       -DTOP_NAMESPACE=glibc -o /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrusers.o -MD -MP -MF /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrusers.o.dt -MT /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xrusers.o
/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xklm_prot.c:73:17: error: implicit declaration of function 'IXDR_GET_iNT'; did you mean 'IXDR_GET_INT32'? [-Werror=implicit-function-declaration]
    objp->svid = IXDR_GET_iNT(buf);
                 ^~~~~~~~~~~~
                 IXDR_GET_INT32
/home/mustafa/İndirilenler/buildroot-ftdev/output/host/bin/arm-buildroot-linux-gnueabihf-gcc /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xspray.c -c -std=gnu11 -fgnu89-inline  -O2 -Wall -Werror -Wundef -Wwrite-strings -fmerge-all-constants -fno-stack-protector -frounding-math -Wstrict-prototypes -Wold-style-definition -fmath-errno    -Wno-unused -fPIC        -I../include -I/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc  -I/home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build  -I../sysdeps/unix/sysv/linux/arm  -I../sysdeps/arm/nptl  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix/arm  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/arm/include -I../sysdeps/arm  -I../sysdeps/wordsize-32  -I../sysdeps/ieee754/flt-32  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I. -nostdinc -isystem /home/mustafa/İndirilenler/buildroot-ftdev/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/7.4.0/include -isystem /home/mustafa/İndirilenler/buildroot-ftdev/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/7.4.0/include-fixed -isystem /home/mustafa/İndirilenler/buildroot-ftdev/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include  -D_LIBC_REENTRANT -include /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/libc-modules.h -DMODULE_NAME=librpcsvc -include ../include/libc-symbols.h       -DTOP_NAMESPACE=glibc -o /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xspray.o -MD -MP -MF /home/mustafa/İndirilenler/buildroot-ftdev/output/build/glibc-glibc-2.28-110-g57922433fa038faa6e37798b9655f85a94978d89/build/sunrpc/xspray.o.dt -MT /home/mustafa/İndirilenler/buildroot-ftdev/output/build/ [... cut ...]
Daha önce python yüklü değil ile alakalı hatalar alıyordum. alias ile python a python3 ü eklemiş ve bashrc üzerinde değişilik yaparak düzelttiğimi sandım ama işe yaramadı. Bu sebeple sudo apt install python yaparak doğrudan kurulum yaptım ve python kurulu değilmiş gibi kurulumu tamamladı. Muhtemelen ya o sürüm yoktu yada doğru bir yol veremediğim için indirdi. Bunu da not düşmüş olayım.
İlk hedefim senin sağlamış olduğun config ile temiz bir derleme sağlamak bunun için yardımcı olur musun ?
Hatanın bir bölümünü içeren kısım hata.txt ile ekte yer almaktadır.
Teşekkür ederim iyi çalışmalar.

furkantokac, 2020-12-02:
Ubuntu 20.04'te sıkıntısız derleme oluyor. Buildroot'un bağımlılıklarını edinmiş olmanız lazım. Yukarıdaki hata farklı bir hatadan kaynaklanıyor olması lazım, asıl hata değil. Buildroot bağımlılıklarıyla alakalı şuradan bilgi alabilirsiniz: https://buildroot.org/downloads/manual/manual.html#requirement

PozitronikTech, 2020-12-02:
Selam Furkan,
Zorunlu Paketleri kontrol ettim.

Paket                 Versiyon
which                    
sed                       4.7
make                    4.2.1
binutils                 2.34-6ubuntu1
build-essential       12.8ubuntu1.1
gcc                       9.3.0
g++                     9.3.0
bash                     5.0.17
patch                    GNU patch 2.7.6
gzip                      1.10
bzip2                     1.0.8
perl                       5.30.0
tar                        1.30
cpio                      2.13
unzip                    6.0
rsync                    3.1.3
file                       5.38
bc                        1.07.1
wget                     1.20.3

python                 3.8.5

Kurulu paketler bunlar. Which için bir kurulum bulamadım. Eksik olabileceğini düşündüğün bir paket var mıdır? Neyi eksik yapıyorum? 

Teşekkür ederim.

furkantokac, 2020-12-02:
Yeni yüklemeler yaptıysanız "output" klasörünü silip tekrar derleme yapıp deneyebilirsiniz. Yine sıkıntı çıkarsa sadece son kısmı değil de baştan sona tüm logu atabilirseniz ona göre bakabiliriz.

PozitronikTech, 2020-12-02:
Output u sildikten sonra yeni bir derleme gerçekleştirdim. 
Gene aynı hatayı aldım. Bu sefer tee ile log dosyası kaydettim. Ek e koydum.

Teşekkür ederim, saygılarımla.
[build.log](https://github.com/furkantokac/buildroot/files/5628301/build.log)

PozitronikTech, 2020-12-03:
Derlemede sıkıntı olup olmadığını anlamak için, hazır da bulunan linux mint yüklü bir makina ile denedim. Gene yukarıda ki hatayı almaya devam ediyorum.
Ubuntu kurulu makinada 20.04 ü kurduktan sonra yaptığım bir iki işlem var.

sudo apt-get install git build-essential libncurses5-dev
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10 

Bu iki olaydan sonra sizin sağlamış olduğunuz derlemeyi gerçekleştiriyorum. Lakin dediğiniz gibi, muhtemelen alakası olmayan yan hatayı görüyoruz. gcc ve g++ kütüphanelerini de şu şekilde kuruyorum.

sudo apt-get install g++-multilib
sudo apt-get install gcc-multilib

Değişen hiçbirşey olmuyor. 
İşletim sistemini kurduktan sonra yapılan farklı bir adım mevcut mu ?
Buildroot un adımlarını takip ettim, istenilen kütüphaneler kurulu gözüküyor.

Teşekkür ederim,  saygılarımla.

furkantokac, 2020-12-03:
build.log dosyası çok yavaş iniyor, galiba Github'tan kaynaklı indiremedim. Farklı bir yerden paylaşabilirseniz iyi olur.

Burada bu repodan ziyade Buildroot'u derlemenizle alakalı sıkıntı var. Ubuntu 20.04 tabanlı bir sistemde daha yeni derleme yaptım, sıkıntı yok. Doğru anahtar kelimelerle ("ubuntu 20.04 buildroot" gibi) sıkıntıyı daha hızlı çözebilirsiniz.

Derlenmiş imajı buradan indirebilirsiniz: https://github.com/furkantokac/buildroot/releases

Qt'yi statik bir şekilde cross-compile yaparak derliyoruz. Yukarıdaki hazır sistem için de bunu yapabilirsiniz fakat kendi sisteminizi derleyerek ardından build-rpi3-qt.sh scriptini çalıştırmanız daha hızlı ve kolay olacaktır. Burada bu işin nasıl yapıldığı adım adım yazıyor: https://github.com/furkantokac/buildroot/blob/ftdev/build-rpi3-qt.sh

PozitronikTech, 2020-12-03:
Teşekkür ederim, deneyeceğim.

furkantokac, 2020-12-04:
Est. ne demek. Sorunuz olursa yazmaya çekinmeyin.

PozitronikTech, 2020-12-07:
Selam Furkan,
Biraz daha araştırmanın sonunda buildroot ile sağlıklı bir derleme yapma
olayını çözdüm. Makina 2011 yılından kalma olduğu için muhtemelen ubuntu
bazı kütüphaneleri yüklemiyor. Bir kaç paket daha buldum ve çözdüm.
Senin config ile sağlıklı bir derleme yaptım. Ayrıca sh dosyası ile de
örnek bir çapraz derlenmiş qt içeren çıktı aldım. Ok işaretleri gayet güzel
bir örnek çıktı olmuş. Eline sağlık. Şimdi asıl soruma geleyim.
Qt creater üzerinde cross compile için bir bölüm var. O şekilde proje
oluşturulabiliyor. Bu elde.
Şimdi senin qt için yaptığın sh dosyasından tek anladığım kısım, derlenecek
yerler gibi config kısımlarının yapılması dosya oluşturma yazma işlemleri,
derleme yap kısımları falan ama şunu anlayamadım.
Ben yaptığım bir qt uygulamasını bu sh a nasıl adapte edeceğim?
Nerede dosyasını oluşturmalıyım?
Dosya ismi ne olmalı?
Bu kısımla ilgili biraz açıklama yapabilir misin?

3 Ara 2020 Per 11:15 tarihinde Furkan Tokaç <notifications@github.com> şunu
yazdı:

furkantokac, 2020-12-07:
Merhaba Mustafa,

Normal Qt projesi nasıl derleniyorsa o şekilde derleme yapılacak, build-rpi3-qt.sh scripti ile bir alakası yok kişisel Qt projesi derleme aşamasının.

1. `cd qtProjeminKlasoru`
2. `buildrootKlasorununKonumu/output/qt/qt-everywhere-src-*/aaaout/bin/qmake`
3. `make`

PozitronikTech, 2020-12-07:
Selam Furkan,
En kısa sürede deneyeceğim teşekkür ederim. Teşekkür ederim.

Saygılarımla.

7 Ara 2020 Pzt 11:22 tarihinde Furkan Tokaç <notifications@github.com> şunu
yazdı:

PozitronikTech, 2020-12-07:
Selam Furkan,
Ekranın sol üstünde qt dosyam. Sağında qt dosyasının uç birimde açılmış
hali. Aşağıda qmake uzantısı. Qmake yapınca - - help açılıyor gibi. Ayrıca
make ile komut yürütülmüyor. Burada benim qt dosyalarımda birşey mi eksik?
Bilmeyene söylüyor gibi söylemen mümkün mü neresini kaçırıyorum?

7 Ara 2020 Pzt 11:22 tarihinde Furkan Tokaç <notifications@github.com> şunu
yazdı:

furkantokac, 2020-12-07:
Yukarıdaki komutların arka arkaya çalıştırılması lazım, yani 2. qmake komutu qt projesinin ana dizininde yani .pro dosyasının olduğu yerde çalıştırılması lazım. Qmake makefile üretir dolayısıyla qmake çalışmazsa make de çalışmaz.

Qt proje dosyası buildroot klasöründeyse tam olarak komutlar şöyle olmalı

```sh
cd buildrootDizini/qtProjesi
../output/qt/qt-everywhere-src-*/aaaout/bin/qmake
make
```

Tercihen build için qtProjesi klasörü içinde yeni bir klasör oluşturmak ana projeyi demiz tutar

```sh
cd buildrootDizini/qtProjesi
mkdir build
cd build
../../output/qt/qt-everywhere-src-*/aaaout/bin/qmake ../
make
```

muratozk, 2020-12-13:
Merhaba Furkan,

Raspberry pi 3 ile kılavuza uygun olarak uyguladım. QML uygulaması da geldi ekrana. Son olarak sh betiğini çalıştırıp QT kurulumunu gerçekleştirip son yorumda yazdığın gibi kendi uygulamamı derledim. Bu noktadan sonra acaba bu kendi uygulamamı senin içinde test amaçlı gönderdiğin uygulama ile nasıl değiştirebilirim ? O kısmı anlayamadım. Derlediğim dosyasının ayrıntıları şu şekilde:
` ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 4.14.0, not stripped`

Ayrıca strip işlemi için sanırım static olarak uygulamayı derlemek gerekiyor. O static derleme konusu hakkında da biraz daha bilgi verebilir misin?

`strip: Unable to recognise the format of the input file `MDPad'`

furkantokac, 2020-12-14:
Merhaba @muratozk ,

Strip işlemi binary dosyanın boyutunu küçültüyor. Yapmasanız da olur fakat yapmak hiç zor değil, "strip binaryAdi" şeklinde kendi uygulamanıza uygulayabilirsiniz.

Oluşturduğunuz imajı hafıza kartına gömdükten sonra hafıza kartını bilgisayara bağlayıp `/root/` klasörü içindeki `QmlTestApplication`'u silip, aynı isimle kendi dosyanızı oraya koyarsanız en kolay şekilde kendi uygulamanızı çalıştırmış olursunuz. Daha detaylı başlangıç işlemi yapmak isterseniz `/etc/init.d/S21ftDev` scriptini oynayabilirsiniz. `QmlTestApplication` uygulaması `/etc/init.d/S21ftDev` scripti tarafından çalıştırılmakta.

muratozk, 2020-12-14:
Merhaba @furkantokac ,
Dediğin gibi yaptım çalıştı :) Çok sağol güzel bir döküman olmuş. Bir kaç sorunlar ve geliştirmesi gereken yerler var. Keşke o kadar bilgim olsa da yardım edebilsem. Mesela açılıştaki ahududu resmini değiştirmeyi nasıl yapabiliriz ? Mesela farklı çözünürlükteki hdmi ekranları taktığımda tuhaf bir ekran karşılılıyor. Çözünürlüğü acaba nasıl değiştirebiliyoruz ki? i2c ve etherneti çalışan olan bir versiyonu da olsa tadından yenmez :) Gerçekten emek kokan bir iş yapmışsın başarılar dilerim.

furkantokac, 2020-12-15:
> Bir kaç sorunlar ve geliştirmesi gereken yerler var.

Bu temel bir dağıtım. Farklı ihtiyaçlara yönelik çalışmaları, bunun üzerine eklemeler yaparak oluşturmak gerekiyor.

> Açılıştaki ahududu resmini değiştirmeyi nasıl yapabiliriz ?

Açılışı değiştirmek için linkteki README.txt dosyasının "Changing boot image"' kısmına bakabilirsiniz.
https://github.com/furkantokac/buildroot/tree/ftdev/board/ftdev/rpi3

> Çözünürlüğü acaba nasıl değiştirebiliyoruz ?

Çözünürlük işi biraz ucu açık bir iş. Nasıl yapılacağını aşağıda belirteceğim, çoğu durumda yeterli olacaktır.
* Programı başlatırken gerekli parametreler ile başlatın. Program `/etc/init.d/S21ftDev` scripti içinde başlıyor. Parametrelerin bazılarını şuradan görüntüleyebilirsiniz: https://doc.qt.io/qt-5/embedded-linux.html Mesela `QT_QPA_EGLFS_PHYSICAL_HEIGHT` ve `QT_QPA_EGLFS_PHYSICAL_WIDTH` parametreleri en çok kullanılan parametreler. Bunlar ile uygulamayı başlatmak için `/etc/init.d/S21ftDev` içindeki `/root/QmlTestApplication --platform eglfs` komutunu `QT_QPA_EGLFS_PHYSICAL_HEIGHT=100 QT_QPA_EGLFS_PHYSICAL_WIDTH=100 /root/QmlTestApplication --platform eglfs` şeklinde değiştirmeniz gerekiyor.
* Yukarıdaki durumu aşan durumlarda kod üzerinden arayüz ayarlamaları yapmanız gerekebilir.
* Bazı durumlarda hafıza kartının `boot` kısmındaki config.txt'ye parametre eklemeniz gerekebilir.

> i2c ve etherneti çalışan olan bir versiyonu da olsa tadından yenmez :)

Diğerki işler için dağıtım ve kernel ayarları yapmak lazım. Ayarlar yapılır fakat test etmek zaman alıyor. Zaman ayırabilirsem dediğiniz özellikleri içinde barındıran hazır Config'ler yapacağım, başkalarına da faydalı olacağını düşünüyorum.

> Başarılar dilerim.

İyi çalışmalar. :)

furkantokac, 2020-12-15:
@muratozk 

> Bir kaç sorunlar ve geliştirmesi gereken yerler var.

Bu temel bir dağıtım. Farklı ihtiyaçlara yönelik çalışmaları, bunun üzerine eklemeler yaparak oluşturmak gerekiyor.

> Açılıştaki ahududu resmini değiştirmeyi nasıl yapabiliriz ?

Açılışı değiştirmek için linkteki README.txt dosyasının "Changing boot image"' kısmına bakabilirsiniz.
https://github.com/furkantokac/buildroot/tree/ftdev/board/ftdev/rpi3

> Çözünürlüğü acaba nasıl değiştirebiliyoruz ?

Çözünürlük işi biraz ucu açık bir iş. Nasıl yapılacağını aşağıda belirteceğim, çoğu durumda yeterli olacaktır.
* Programı başlatırken gerekli parametreler ile başlatın. Program `/etc/init.d/S21ftDev` scripti içinde başlıyor. Parametrelerin bazılarını şuradan görüntüleyebilirsiniz: https://doc.qt.io/qt-5/embedded-linux.html Mesela `QT_QPA_EGLFS_PHYSICAL_HEIGHT` ve `QT_QPA_EGLFS_PHYSICAL_WIDTH` parametreleri en çok kullanılan parametreler. Bunlar ile uygulamayı başlatmak için `/etc/init.d/S21ftDev` içindeki `/root/QmlTestApplication --platform eglfs` komutunu `QT_QPA_EGLFS_PHYSICAL_HEIGHT=100 QT_QPA_EGLFS_PHYSICAL_WIDTH=100 /root/QmlTestApplication --platform eglfs` şeklinde değiştirmeniz gerekiyor.
* Yukarıdaki durumu aşan durumlarda kod üzerinden arayüz ayarlamaları yapmanız gerekebilir.
* Bazı durumlarda hafıza kartının `boot` kısmındaki config.txt'ye parametre eklemeniz gerekebilir.

> i2c ve etherneti çalışan olan bir versiyonu da olsa tadından yenmez :)

Diğerki işler için dağıtım ve kernel ayarları yapmak lazım. Ayarlar yapılır fakat test etmek zaman alıyor. Zaman ayırabilirsem dediğiniz özellikleri içinde barındıran hazır Config'ler yapacağım, başkalarına da faydalı olacağını düşünüyorum.

> Başarılar dilerim.

İyi çalışmalar. :)

closed 2020-12-15 by furkantokac