furkantokac/buildroot #14: How to enable uart for debugging on usb (usb-ttl)

furkantokac, 2020-12-16:
## To enable ttyAMA0 for debugging

* Put `8250.nr_uarts=1` to cmdline.txt in boot partition.

## To redirect the console to ttyAMA0;

1. Go to /etc/inittab
2. Change the line `console::respawn:/sbin/getty -L  console 0 vt100 # GENERIC_SERIAL`
  to `ttyAMA0::respawn:/sbin/getty -L  ttyAMA0 0 vt100 # GENERIC_SERIAL`

## How to do physical connection

1. Get your 3.3v usb-ttl hardware.
  [image]
2. USB-TTL's TX to RPI's RX which is pin10 (GPIO15)
  USB-TTL's RX to RPI's TX which is pin8 (GPIO14)
  USB-TTL's ground to RPI's ground which is pin6 (Ground)
  [image]

closed 2020-12-16 by furkantokac

michal-lecki, 2021-01-12:
Hello @furkantokac.
I am trying to enable console access as you described above, but I am still getting only this output on UART:
`
Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... done.
QIconvCodec::convertToUnicode: using Latin-1 for conversion, iconv_open failed
QIconvCodec::convertFromUnicode: using Latin-1 for conversion, iconv_open failed
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QFbVtHandler: socketpair() failed (Function not implemented)
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
libpng warning: iCCP: known incorrect sRGB profile
`

furkantokac, 2021-01-12:
> Hello @furkantokac.
> I am trying to enable console access as you described above, but I am still getting only this output on UART:
> `Starting syslogd: OK Starting klogd: OK Initializing random number [...]

You're successfully done it, just app blocks the terminal login. Just go to `/etc/init.d/s12ftdev` and comment everything in it, or start the app in background by adding `&` the the end of the line of the app starting command.

michal-lecki, 2021-01-13:
@furkantokac  Thank you for fast response. It helped, but I also had a typo in /etc/inittab which took me some time.

Works like a charm now.