furkantokac/buildroot #15: 404 not found when running `make -j8`

hysan, 2020-12-17:
Hi, thanks for putting this together. I'm trying to create an image with USB (and maybe wifi), but when I try to run `make -j8`, I end up getting a 404 not found on this url: http://sources.buildroot.net/linux-c117a8bccf37bfba323065b566cf999ed4629a4a.tar.gz

I've been trying to figure out why this is the case, but since it's the first time for me to do anything this low level, I'm coming up empty.

Also, I saw in a [previous issue](https://github.com/furkantokac/buildroot/issues/1#issuecomment-636375555) that you were considering making a fastboot + USB + Networking image. Did you ever find time to do that?

furkantokac, 2020-12-17:
I have just checked and there is no problem. You shouldn't got any problem when you run the build without changing any option like `make ftdev_rpi3_fastboot_defconfig` and `make`.

hysan, 2020-12-18:
I rebooted my computer, deleted everything, reinstalled dependencies, and pulled down a fresh copy of the repo, and it works now! Thank you for doing a sanity check for me. I must have had something wrong on my system.

closed 2020-12-18 by hysan

furkantokac, 2020-12-18:
Thats nice, good job!