furkantokac/buildroot #16: A few errors

ucondua, 2021-01-07:
make[3]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase/src/network»
make[3]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase/src/gui»
make[2]: *** [Makefile:569: sub-gui-install_subtargets] Ошибка 2
make[2]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase/src»
make[1]: *** [Makefile:63: sub-src-install_subtargets] Ошибка 2
make[1]: выход из каталога «/home/reanon/projects/test_kernel/buildroot-2020.02.2/output/qt/qt-everywhere-src-5.14.2/aaabuild/qtbase»
make: *** [Makefile:100: module-qtbase-install_subtargets] Ошибка 2
Help me please ))

furkantokac, 2021-01-07:
Did you enable Qt from the `make xconfig` ?

ucondua, 2021-01-07:
nconfig

furkantokac, 2021-01-07:
Don't enable Qt by the config. Just compile your system, and run the ./build-rpi3-qt.sh script.

ucondua, 2021-01-08:
Hey.  It compiled very well.  But I need to remotely run a QT application.
I will include usb and wifi support in the kernel.  Can't configure Qt
startup kit yet. Thank you.

furkantokac, 2021-02-14:
Please check previously closed-issues for the wifi and usb support.

closed 2021-02-14 by furkantokac