furkantokac/buildroot #17: Blank screen before Qt App

michal-lecki, 2021-02-26:
Hello @furkantokac.
Once again - big thank you for sharing this project.

I am struggling with an issue of visible console before QT takes over in my app.
I am powering on my screen in the same time as RPi (3B+), and in some cases of power cycling (on-off-on-off in different time intervals) my screen turns on faster than usual.

I would like to make screen ON only when QT has already taken over framebuffer.

One solution is to power up screen only after app starts, but this I would rather like to avoid - electrical design change + longer overall boot time.
I am looking for different, software approach, but could not find any proper reference on how to achieve my goal.

Maybe disabling console at all? Maybe some call to HDMI driver?

Do you have any suggestions, or maybe can point me to some reading?

Thank you in advance.

furkantokac, 2021-02-27:
I see your point. This is about display driver chip > RPI > Linux. There can be lots of different strategies to solve this problem. The best and un-hacky way is trying to control the display power by RPI. But first, let's identify the system;

1. How you connect to display, by HDMI or GPIO pins ?
2. How you power your display, by USB or externally ?
3. What is the model of display driver chip ? Reading the datasheet of it can be very helpful.

michal-lecki, 2021-02-27:
1.HDMI
2.External power.
3.It's Chinese HDMI-DSI board to Chinese 8" oval screen. Technical support and data sheet is a bad joke. It took me two months of messaging to get screen timing data to configure Rpi video output. Even then I got a screenshot from other documentation. But it worked.
I can change my electrical design to powerup display whenever I like, but it would add to overall boot time.

furkantokac, 2021-05-17:
> I can change my electrical design to powerup display whenever I like, but it would add to overall boot time.

Why ? It should not have a significant impact on boot time. You can embed display-power-up command into your application easily.

closed 2021-05-18 by furkantokac