furkantokac/buildroot #18: Problem with cross-copmiplation of QT

krzychk2, 2021-03-26:
Hi. 
I have a problem with cross-compilation of QT using Your script: build-rpi3-qt.sh

QT downoladed version is 5.12.1

I have error:
ERROR: Feature 'opengles2' was enabled, but the pre-condition 'config.win32 || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.

ERROR: The OpenGL functionality tests failed!

You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],

QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

make: *** Nie podano obiektów i nie znaleziono makefile. Stop.

make: *** Brak reguł do wykonania obiektu 'install'. Stop.

I have looked in different forum and it looks like problem with opengl libraries paths for RPi. 

Also I've got suggestion to add if - downloaded file statent to script, just to avoid downloading already downoladed qt source-code.
Thanks.

furkantokac, 2021-03-28:
Hey @krzychk2 

Did you run `make ftdev_rpi3_fastboot_defconfig` or just run `build-rpi3-qt.sh` ? If the buildroot compile without any problem at first, Qt should be built without any problem. It is tested.

> Also I've got suggestion to add if - downloaded file statent to script, just to avoid downloading already downoladed qt source-code.

Thanks for your suggestion. I thought that putting a simple script and let people easily play on it is more handy than having a complex script.

krzychk2, 2021-03-29:
Hey @furkantokac 

Yes, I have ran make ftdev_rpi3_fastboot_defconfig and it has built without any problems. Maybe You have something in Your enviromental settings that builds the Qt without any problems.

furkantokac, 2021-05-17:
Hey @krzychk2 

I always use this build on different PCs. I have just build it for you, and I verify that it is working without any problem. My environment: KDE Neon Unstable (based on Ubuntu 20.04) 

By checking the log you shared, I suspect that there is a problem on your Buildroot build. Please be sure that you've installed all the Buildroot dependencies and build your distro without any problem.

krzychk2, 2021-05-18:
Hey @furkantokac
Well, I don't know what happened but I have cleaned the project, rebuilt it and all worked. qmake has build. Script has an error at the end so built qmake does not move to the final folder, where the script claims it should be - but all is working. 
Thanks fot the support

closed 2021-05-18 by krzychk2

furkantokac, 2021-05-18:
@krzychk2 As I've though, I think that you've build the distro without cleaning your old distro in the `buildroot/output` folder and it created a problem. In Buildroot, I recommend you that remove the `buildroot/output` folder if you go to a big update, package removal or kernel update on your current build. By practicing, you'll understand when you need to remove the `buildroot/output` folder. Also, create an independent download folder to use all of your buildroot projects, as I did in my config `BR2_DL_DIR="$(HOME)/.dl"` so you'll save lots of time with that.