furkantokac/buildroot #20: Testing your examples.

insistrain, 2021-07-07:
I'm running ubuntu on a MAC, usign VMware Fusion, so it's a virtual ubuntu machine, also I've a SDcard USB adapter to get  mount the cards.
Every time I want to access the SD files in the ROOT partition it's  Read Only Filesystem. I can' do nothing to modify a file there. So, I can't apply your suggest in another issue: Change the line console::respawn:/sbin/getty -L console 0 vt100 # GENERIC_SERIAL
to ttyAMA0::respawn:/sbin/getty -L ttyAMA0 0 vt100 # GENERIC_SERIAL

Neither as superuser, or root user.
Maybe it is because I use a sd adapter ? Or blame the virtual machine ... I don't know.

To check a fast boot and run my own code, first,  i will try as you suggested other guys, the simple HelloWorld app. That's a very good idea. !!

Thank you & Cheers !
Rick.

insistrain, 2021-07-07:
Finally I was able to modify the inittab file in the card ! (I should use another PC with ubuntu).
Now when I boot the pi with your example, it only shows arrows images rotating on the screen, and by serial I get this messages below....
Why I can't see another image on my monitor ? Is is normal ?

Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... done.
QIconvCodec::convertToUnicode: using Latin-1 for conversion, iconv_open failed
QIconvCodec::convertFromUnicode: using Latin-1 for conversion, iconv_open failed
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QFbVtHandler: socketpair() failed (Function not implemented)
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
libpng warning: iCCP: known incorrect sRGB profile

furkantokac, 2021-07-07:
* To see the terminal instead of rotating arrows, check this answer: https://github.com/furkantokac/buildroot/issues/14#issuecomment-759085963

* To integrate your application, I recommend you that statically compile all of your app and dependencies and replace it with the default Qt application. If your application doesn't rely on some specific kernel features etc, it should just work fine without any extra effort.

insistrain, 2021-07-07:
Thank you Furkan !!
Now I'll about the terminal problem, and then, realize a simple "HelloWorld" program to let it work instead your app.

insistrain, 2021-07-07:
Ok, at this point I've the terminal working fine, and in it, I can see the board booting AND the HelloWorld text  also! So it's a success!
Now is coming the "hard" part to get my real app working after a short time boot....
Maybe it takes me a lot of time, since statically compile it, seems to be a nightmare with my poor knowledge.

Anyway, again thank you so much for your support Furkan. 
Cheers.
Rick.

furkantokac, 2021-07-07:
> Ok, at this point I've the terminal working fine, and in it, I can see the board booting AND the HelloWorld text also! So it's a success!

Nice!

> Maybe it takes me a lot of time, since statically compile it, seems to be a nightmare with my poor knowledge.

https://github.com/furkantokac/buildroot/blob/ftdev/build-rpi3-qt.sh This script statically compiles the Qt. You can use this as a reference. `FT_CROSS_COMPILE` and `FT_SYSROOT` what you'll need to. As I said before, if your app is not rely on some kernel features or graphical interface, it should not be that hard. 

You'll do it. Keep going!

closed 2021-07-07 by furkantokac

insistrain, 2021-07-07:
Dear Furkan.
Magically I've got compile my app statically much faster as I hope. That's great for me.
I've tested the app with another raspian PI card, and works fine in static way, but.
When I try to get it work with the card generated by the "fastboot", I get the following error in the serial, and also can't see any graphics on screen, only the four raspberry images:

Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... done.
/root/a.out: error while loading shared libraries: libasound.so.2: cannot open shared object file
: No such file or directory

Obviously the name of the app is a.out, and I never use that 'libasound' lib in the code ???
Is is a dependency of the main library ? 
Now, how to solve this so ?

Cheers!
Rick.