furkantokac/buildroot #21: Can't get my app work (Not a Qt app).

insistrain, 2021-07-07:
Dear Furkan.
Magically I've got compile my app statically much faster as I hope. That's great for me.
I've tested the app with another raspian PI card, and works fine in static way.  (but in this raspberry the libasound.so.2 lib is installed).
When I try to get it work with the card generated by your "fastboot", I get the following error in the serial, and also can't see any graphics on screen, only the four raspberry images:
My app use graphics, and only show png images moving on screen. I used SDL2 library.

Here's what i get in the serial console:

Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... done.
/root/a.out: error while loading shared libraries: libasound.so.2: cannot open shared object file
: No such file or directory

Obviously the name of the app is a.out, and I never use that 'libasound' lib in the code ???
Is is a dependency of the main library ? 
Now, how to solve this so ?

Cheers!
Rick.

insistrain, 2021-07-08:
Hi.
I've tried adding Alsa Utils in your fastboot config, and compile it again. But now I don't have another dependency... Uart msg say:

Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... done.
/root/a.out: error while loading shared libraries: libpulse-simple.so.0: cannot open shared object file: No such
 file or directory

So i guess i have not well  statically compile my app, but don't know how to do it, because libasound.so.2 & libpulse-simple.so.O doesn't have the ".a" files to get statically compile them. :(

To compile my SDL2 app I used:
gcc -Wall image_test.cpp /usr/lib/arm-linux-gnueabihf/libSDL2.a -Wl,--no-undefined -lm -ldl -lasound -lm -ldl -lpthread -lpulse-simple -lpulse -lsndio -Wl,-rpath,/opt/vc/lib -L/opt/vc/lib -lbcm_host -lX11 -lXext -lXcursor -lXinerama -lXi -lXrandr -lXss -lXxf86vm -lwayland-egl -lwayland-client -lwayland-cursor -lxkbcommon -lpthread -lrt

Just get it googling...
Ideas ?
Thank you.

insistrain, 2021-07-09:
Hi.
I was googling a lot, trying with different compilations options of my code,  without luck.
At this point I'm very frustrated.
Just think in add to the buildroot config the SDL2 lib, and also I add X11 support, generate again the sdimage and test it.
Although the app is not statically compiled, the booting times were not affected at single view, well. And it is usefull for me anyway, but.
Now I've a different error in terminal.
This is:

Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... done.
SDL could not initialize! SDL_Error: No available video device
Failed to initialize!

Obviously there is some missing, but seems to be better than before, at least no libraries missing.
Ideas ?

Cheers.
Rick.

insistrain, 2021-07-13:
Ok at this time and after a long research and tries, I've the graphical part of an app working with fastboot !!
Like I was not able to compile my app statically, i've add the SDL2 and all of the dependencies required by it, in the "make menuconfig" part and i generate again the sdcard.image.
Now I've tried with the real app that only includes a serial port that uses "wiringpi" to work.
And unfortunately have a new error msg: "Falied to initialize SERIAL!"

In cmdline I've:
root=/dev/mmcblk0p2 rootwait console=tty1 vt.global_cursor_default=0 8250.nr_uarts=1

and in config.txt:
#---> Boot
kernel=zImagen
disable_overscan=1
boot_delay=0
enable_uart=1
disable_splash=1
avoid_warnings=1
dtoverlay=pi3-miniuart-bt

#gpio
gpio=2=op,dh

#video
hdmi_group=2
hdmi_mode=16

Please, ideas ??
Cheers.
Rick.

insistrain, 2021-07-14:
Hi.
Well, I've all working fine now !!
The last issue talking about serial, using wiringpi was solved like this:
In my app when I open the serial port I used:
    if ((fd = serialOpen ("/dev/serial0", 115200)) < 0)
	{
	   printf( "Failed to initialize SERIAL!\n" );
	   return 1 ;
	}
That's why I get that error.... was written in the app.
But, with naming "/dev/serial0" was working on a raspberry pi 3+ with a lite raspian OS. And not with the fastboot config.

I changed it by "/dev/ttyAMA0" and try again in my board using the fastboot config, and the error msg was out, BUT, the serial port won't receive or transmit, anything.

So I changed it again by "/dev/ttyS0" and surprisely for me, my board begin to work fine usign the fastboot config, and the error msg  goes out too.

I could not say why it's working now..... I'm very begginer in raspberry, C++, and with buildroot too.
Also, I'm an self student, and a old man :) 

THANK YOU Furkan !!!  You did a great job with the fastboot example.!!
Cheers.
Rick.

furkantokac, 2021-07-19:
Great work! Thanks for your detailed explanation. This will help to other devs.

closed 2021-07-19 by furkantokac