furkantokac/buildroot #24: Cannot get new package (QT Module) to work

DanielVolt, 2021-07-16:
Hi,

When building buildroot with added package (qtquicktimeline) using make xconfig the resulting qt app will error out saying: "module qtquicktimeline not found".

How do we properly add the qt module?

Many thanks

DanielVolt, 2021-07-19:
any help??

furkantokac, 2021-07-19:
Hey Daniel,

The way I recommend you to build your Qt project is that

1. Build your distro without any Qt package
2. Statically cross-compile Qt from the source for your distro (you can use the script, please read the readme)
3. Statically compile your Qt app by the Qt you've compiled. You'll have a single executable file on this stage.
4. Put the exec file to the RPI and run.

DanielVolt, 2021-07-20:
Hey Furkan,

Thank you so much! Since other devs will most likely need this to here is the full instructions on how to add QT Modules:

1. git clone a QT Module in the buildroot -> output -> qt -> qtfolder (/buildroot/output/qt/qt-everywhere-src-5.12.1):
- "git clone git://code.qt.io/qt/qtquicktimeline.git -b 1.0"
2. cd to the qt module dir:
- "cd qtquicktimeline/"
3. run qmake on qt module:
- "/buildroot/output/qt/qt-everywhere-src-*/aaaout/bin/qmake"
4. run make on qt module
- "make"
5. run make install on qt module to install said module
- "make install"

and tada! qt module ready to be used, re-compile your app and boom.

Teşekkür ederim Furkan jan :-)

closed 2021-07-20 by DanielVolt