furkantokac/buildroot #26: Cannot run script

a-matos00, 2021-07-24:
I have sucessfuly created the SD card image, but when I run "./build-rpi3-qt.sh" in terminal it says : "bash: ./build-rpi3-qt.sh: No such file or directory".

Is this automaticaly generated? I can't seem to find it anywhere.

furkantokac, 2021-09-17:
`build-rpi3-qt.sh` is on the root folder path https://github.com/furkantokac/buildroot/blob/ftdev/build-rpi3-qt.sh

So just;

1. Build your image
2. Go to the root folder, which should be named as "buildroot" if you didn't change it
3. Run the script by `./build-rpi3-qt.sh`

a-matos00, 2021-09-17:
Thanks for the reply, for some strange reason the script was not in the buildroot folder. So I added it manually and now I can run it. Issue solved! Cheers.

closed 2021-09-17 by a-matos00