furkantokac/buildroot #27: Help for start a video after booting

l0l0l0, 2021-09-16:
Hello,
Already great job thank you.
I am on this project: [https://withrow.io/simpsons-tv-build-guide](url) and I want to launch a video after booting the raspberry.
I found this that could do the trick but it looks old and complicated to me:
[https://thebugfreeblog.blogspot.com/p/build-procedure-for-piomxtextures.html](url)
[https://github.com/carlonluca/pot](url)
maybe you have a simpler solution?
Thank you in advance for your help

furkantokac, 2021-09-17:
If you just need to run a video, the simplest thing that comes in my mind is;

1. adding gstreamer libraries to your buildroot image
2. adding a simple player to your buildroot image
3. triggering the video on the startup

To get help about the gstreamer libraries that you'll need, check this config: https://github.com/furkantokac/buildroot/blob/tk1/br2-external-ftdev/configs/ftdev_tk1_defconfig This config is prepared to run ogg sound files from a qt application. Also the second option is, which will give you much more flexibility is;

1. adding gstreamer libraries to the buildroot image (qt use this 
2. building the qt statically from the source (you can use the script: https://github.com/furkantokac/buildroot/blob/ftdev/build-rpi3-qt.sh )
3. managing your video by qt application that you write (you can develop your application on your PC at this point)
4. statically compiling your qt application with the qt that you will be built on the step 2
5. running your statically compiled qt application on the startup

If you've some experience about the topics, you can do it. It's not that hard.

l0l0l0, 2021-09-21:
Great, thanks for your reply.

closed 2021-09-21 by l0l0l0