furkantokac/buildroot #28: Can I build fast boot for rpi4(rpi4 module)

Vd2617, 2021-09-29:
Hi. Thank you for you fast boot build it work very well(i use it with can0) for a long time. Can I build it for rpi4 (rpi4 module) ? Please tell me the algorithm of actions for this?Thank you.

furkantokac, 2021-11-08:
Hey @Vd2617 

This work is very dependent to the hardware, that's why it requires a detailed investigation. The logic behind the fastboot optimization is same. You can check my blogpost and follow the steps. While moving forward, you'll face with some specific problems and by solving them, you can move on to the next step.

I would like to work on the RPI4 too but I don't have time for that. Maybe I'll work on it in future.

Vd2617, 2021-12-09:
I can build system for rpi4modulle with your kernel settings.(disable in kernel) . But i do not add device-tree to kernel(because easy add new overlays without compile device-tree when device tree not i ).How much time try to  load os when device-tree is not in kernel?

furkantokac, 2021-12-09:
This is not directly "device tree takes the time" . Check the "**K2 - Linux pre-boot stage**"  part in the blogpost: https://furkantokac.com/rpi3-fast-boot-less-than-2-seconds/ You can see the how much time you save by embedding device tree to the kernel. It's a very big save.

Also, the blogspost is valid for RPI3, I didn't work on the RPI4. But, you're on the right way. Keep pushing!

Vd2617, 2021-12-09:
I am build without device-tree.All work. As i understand ,when i see color rectangle it means that gpu test good and start load kernel? My timeline from color rectangle to load  login root 2 sec. Can you help me , how  build static qt 15.5.2 for my os , path to my sysroot , and my compiler in your script and all work ?

Vd2617, 2021-12-12:
I get problem with vc4 kernel module,if i use make linux-xconfig and disable functions (wich you used) , i have not module vc4 in target /lib/kernrl/gpu/drm/

Vd2617, 2022-01-04:
Can you help me please. I am  try to build for rpi4 . My steps ->  (1)make rpi_default_deconfig - > (2)make menuconfig -> (3)make xconfig(add vc4 , mesa3d and e.t.c) -> (4)make linux-xconfig(in it i try to disable you settings hw_random and e.t.c) -> (5)make -j16 . But when I do it ,the step (3) not save and system build without step(3) settings , and that i can not see vc4 module which i am add in step(3). When i am not use step(4) settings in step(3) work but whithout step(4) i can not disable your options and not get fast boot ...Help me please

Vd2617, 2022-01-04:
I am think that , I misunderstand how  do steps to build