furkantokac/buildroot #29: "make linux-xconfig" command not working

a-matos00, 2021-10-02:
I am trying to enable the CAN interface by following this thread https://github.com/furkantokac/buildroot/issues/10. I managed to make xconfig, but when I run "make linux-xconfig" I get the following output

/bin/make -j1 O=/home/andrija/buildroot-ftdev/output HOSTCC="/bin/gcc" HOSTCXX="/bin/g++" syncconfig
BR_BINARIES_DIR=/home/andrija/buildroot-ftdev/output/images PKG_CONFIG_PATH="" /bin/make -j5 -C /home/andrija/buildroot-ftdev/output/build/linux-c117a8bccf37bfba323065b566cf999ed4629a4a HOSTCC=/bin/gcc HOSTCC="/bin/gcc -O2 -I/home/andrija/buildroot-ftdev/output/host/include -L/home/andrija/buildroot-ftdev/output/host/lib -Wl,-rpath,/home/andrija/buildroot-ftdev/output/host/lib" ARCH=arm INSTALL_MOD_PATH=/home/andrija/buildroot-ftdev/output/target CROSS_COMPILE="/home/andrija/buildroot-ftdev/output/host/bin/arm-buildroot-linux-gnueabihf-" DEPMOD=/home/andrija/buildroot-ftdev/output/host/sbin/depmod INSTALL_MOD_STRIP=1 HOSTCC="/bin/gcc" xconfig
  MOC     scripts/kconfig/qconf.moc
/bin/sh: 1: /home/qt/work/install/bin/moc: not found
make[3]: *** [scripts/kconfig/Makefile:292: scripts/kconfig/qconf.moc] Error 127
make[2]: *** [Makefile:517: xconfig] Error 2
make[1]: *** [linux/linux.mk:565: /home/andrija/buildroot-ftdev/output/build/linux-c117a8bccf37bfba323065b566cf999ed4629a4a/.kconfig_editor_xconfig] Error 2
make: *** [Makefile:84: _all] Error 2

Obviously the /home/qt/work/install/bin/moc file is missing. Any hints on how to obtain it? If you need additional information I will provide it.

a-matos00, 2021-10-02:
Or is there maybe any workaround for me to enable
 - NET
- CAN
- CAN_DEV
- SPI
- CAN_MCP251X
settings without using linux-xconfig?

a-matos00, 2021-10-02:
I managed to run the command after repeating the make process without any changes. Still not sure what was the problem.

l0l0l0, 2021-10-04:
make linux-menuconfig does the same with another interface

a-matos00, 2021-10-05:
I think my issue was caused by me not having qt5 dev tools installed. I installed them after the initial build to get make xconfig working. But running linux-xconfig required a full rebuild with qt5 dev tools installed.