furkantokac/buildroot #34: Can i Run this script on armbian os and orange pi +2E?

EmadDeve20, 2021-11-03:
Hi. I developed a skid loader and I have to use this script to run!
And I have to use a USB flash drive for the user! Because a user may want to listen to music.
Anyway!

and after this command: ```make -h8``` I got Error with this message: ``` You must install `python` on your build machine```!

I think I can solve this error by creating an alias in bashrc file like This:
```alias python="python3"```

but this does not work!
how can I fix it?

python path in script just for rasbian os on rasbery pi 3 right?
and I must changing my python path in script to my python path! i right?

ltorsini, 2022-09-01:
Same issue: You must install `python` on your build machine

EmadDeve20, 2022-09-01:
Hello @ltorsini  i can't remember why i open this issue!
I so sorry :)
Because I Work in lot of project and when i opened this issue,
I worked a project like your repo. and past
for this i can't rememeber why I open this issue.