furkantokac/buildroot #38: USB mouse/kb working in console but not Qt app

joshellissh, 2022-07-21:
First of all, great work on the repo/config! It's made my RPi3 boot as fast as I need it, which is great!

I am having one issue, and while I know it's probably not directly related to buildroot, I was wondering if anyone seeing this might have any ideas (as I might be related to Qt build config? Not sure).

Anyway, I was able to add the necessary HID/USB settings back to the kernel config to enable my USB keyboard and mouse. They get power now, and the keyboard works in the console as does the mouse (I can tell using `cat /dev/input/mouse0 | hexdump`). The issue I'm having is when my fullscreen Qt app starts (built using the qt static compiled in this repo) the mouse and keyboard don't work there. 

I know it's a longshot posting that question here, but I was hoping someone might have some insight.

Thanks!

joshellissh, 2022-08-27:
I still haven't had any luck on this. The keyboard and mouse work in the console, but not in Qt. If anyone can help me figure this out I'd be happy to pay you for your time!

a-matos00, 2022-09-26:
Hi,

How did you enable back disabled USB settings?

joshellissh, 2022-09-26:
> Hi,
> 
> How did you enable back disabled USB settings?

Here is the repo I forked with my changes: https://github.com/joshellissh/PiDGC/tree/main/buildroot