furkantokac/buildroot #40: How to change QT app?

SinanAkkoyun, 2022-09-27:
Hey, thank you very much for the repo.
How can I change the QT application on the RPI? I've never worked with QT and would like to replace the spinning arrow example with my own code (not live link)

Thank you very much!

a-matos00, 2022-09-28:
The application is located in the root folder. You need to setup a toolchain for compiling the qt application for this specific linux OS inside Qt Creator. After you compile the OS image. run the qt script in the buildroot folder. Then inside Qt Creator go to tools->kits. Create a new kit. You have to select a QT version(that is your qmake file that is generated after running the script). The gcc and g++ compilers are located in the output/host folders. This will help you get started. There is more info on the internet. Good luck!

SinanAkkoyun, 2022-09-30:
Thank you so much!!!
I did all the above (and added sysroot output/host/arm-buildroot-linux-gnueabihf/sysroot), but I get an error while trying to compile in QT Creator:
`cannot find -lqxcb`

What can I do?

SinanAkkoyun, 2022-10-02:
```rpi-qt-fastboot/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/7.4.0/../../../../arm-buildroot-linux-gnueabihf/bin/ld: cannot find -lqxcb
collect2: error: ld returned 1 exit status
make: *** [Makefile:300: backupQmlTestApplication] Error 1```

a-matos00, 2022-10-02:
I think you are missing that library on your host computer, try googling how to install it.

SinanAkkoyun, 2022-10-02:
> I think you are missing that library on your host computer, try googling how to install it.

I did install every possible variant of xcb packages but it still wont work. The sysroot is missing the package isnt it? How can I disable xcb completely? It must have worked for the author of this repo

SinanAkkoyun, 2022-10-02:
I would love some further help or any advised direction I could head, I tried so much but nothing seems to work...

tjitra, 2025-01-08:
Sory im new with buildrood and qt.
Were folder is the sample qt script apps like on your youtube.

furkantokac, 2025-01-09:
You need to statically compile your Qt app. This repo is also have necessary script that I wrote to compile Qt source code. Check https://github.com/furkantokac/buildroot?tab=readme-ov-file#build-rpi3-fastboot-distro--static-qt 

Just run Buildroot successfully, then compile your Qt project by the qmake located in `./output/qt/qt-everywhere-src-*/aaaout/bin/qmake` folder, and replace the QmlTestApplication here by your application, `https://github.com/furkantokac/buildroot/tree/17be2a2de29cf062decb40f492ce5359862aa829/board/ftdev/rpi3/rootfs_overlay/root` and to apply the file change, re-run the Buildroot.