furkantokac/buildroot #41: libfakeroot issue.

SquirrelTactics, 2023-02-20:
Hello I have been chasing this issue and I can not figure it out or come close to fixing it, I have spent weeks googling and trying to resolve it but it always fails at this spot. please any information would be greatly appreciated.        

|                        struct stat *
libfakeroot.c:89:46: note: in definition of macro 'SEND_GET_STAT64'
   89 | #define SEND_GET_STAT64(a,b) send_get_stat64(a)
      |                                              ^
In file included from libfakeroot.c:60:
communicate.h:209:44: note: expected 'struct stat64 *' but argument is of type 'struct stat *'
  209 | extern void send_get_stat64(struct stat64 *buf);
      |                             ~~~~~~~~~~~~~~~^~~
make[4]: *** [Makefile:642: libfakeroot.lo] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:660: all-recursive] Error 1
make[2]: *** [Makefile:434: all] Error 2
make[1]: *** [package/pkg-generic.mk:241: /home/justin/Desktop/buildroot-ftdev/output/build/host-fakeroot-1.20.2/.stamp_built] Error 2
make: *** [Makefile:84: _all] Error 2

SquirrelTactics, 2023-03-16:
Pretty sure the fix for this is to copy over all the configs to a updated buildroot repository I've got it to build this way and it is the 2sec boot time. the only other part was the building of the qt and copying it to the buildroot I have been unsuccessful in this but it's a separate issue.

closed 2023-03-16 by SquirrelTactics

thefrankk, 2024-03-21:
Witch version of Buildroot did you use when you update ?