furkantokac/buildroot #42: Building issue on WSL

vishesh-varma, 2023-02-21:
Hi! Thank you for this amazing resource! I've been trying to build it as mentioned in README.md, using WSL, but I get the following error when running the command: `make -j8`

```
make[3]: *** [Makefile:642: libfakeroot.lo] Error 1
make[3]: Leaving directory '/home/visheshv/bruh/buildroot/output/build/host-fakeroot-1.20.2'
make[2]: *** [Makefile:660: all-recursive] Error 1
make[2]: Leaving directory '/home/visheshv/bruh/buildroot/output/build/host-fakeroot-1.20.2'
make[1]: *** [Makefile:434: all] Error 2
make[1]: Leaving directory '/home/visheshv/bruh/buildroot/output/build/host-fakeroot-1.20.2'
make: *** [package/pkg-generic.mk:241: /home/visheshv/bruh/buildroot/output/build/host-fakeroot-1.20.2/.stamp_built] Error 2
```

SquirrelTactics, 2023-02-21:
I get the same issue on Ubuntu laptop.

SquirrelTactics, 2023-02-22:
The new master branch calls for fakeroot 1.30.1, pretty sure this one calls for 1.20.2 wonder if it’s superseded and that’s why we get the error. I’m no expert by any means I’ve been dealing with the same issue for weeks

SquirrelTactics, 2023-02-22:
So I just tried that and it fixed it but now gets stuck at c-stack so i guess more superseded packages to fix hope this helps you.

vishesh-varma, 2023-02-24:
I see! Does buildroot not have some sort of package/dependency manager? Because it could take days just to fix all of the superseded packages

SquirrelTactics, 2023-02-24:
I have zero idea. I am pretty sure they are just a set of qmake files that call out to download the packages.

SquirrelTactics, 2023-02-24:
I do have another idea. In the package folder for fakeroot there are two patch files in there and they patch the exact issue we are having. Maybe they are no longer needed

SquirrelTactics, 2023-03-03:
so I have found if you copy the board configs and defconfig over to a new version of buildroot it builds the image.

SquirrelTactics, 2023-03-22:
since you didn't build the config from that repo you have to move a few additional files like bootcode.bin and start.elf to the appropriate folders open the board config and you will see two batch file .sh check what they are pulling and make sure the files are there. it worked for me I have a 2sec boot image Now I'm just having an issue with pulling the qt package into the image. I've never used QT and I'm not a software developer so it's new stuff to learn as I go.

SquirrelTactics, 2023-03-22:
So I did build a new Debian 11 vm and installed all the dependencies for buildroot including all the fetching ones and I had a successful build without using a new repo or anything.  You also need to make sure your gcc version is 9 or lower it has issues with newer ones.

SquirrelTactics, 2023-03-23:
so I have done test and that is the issue make sure you use GCC-9 and not 10 there are a few pages online to help with it cause you have to install both and make 9 the priority. I was using debian 11 for this and everything went well using it. I tried Ubuntu in the past but it didn't work as they are on GCC-11 and even selecting the 9 as priority I had issues.

axon-11, 2023-12-14:
Do you mean setting gcc9 as the priority and then make. if that then it still shows error of libfakeroot.c.
git clone https://github.com/furkantokac/buildroot
cd buildroot
make ftdev_rpi3_fastboot_defconfig
make

Can you please explain. @SquirrelTactics

axon-11, 2023-12-15:
I see a number of bootcode.bin files in the old. also the same are present in the new build. what should i exactly transfer

SquirrelTactics, 2023-12-26:
Yes sorry it's been so long since i have messed with this stuff but using Debian 11 and setting GCC9 as the priority resolved the issues I was having. I didn't even have to transfer anything I was about to use the original build repo.

tjitra, 2025-01-28:
Im using debian 12, i try install gcc 9 but always fail with some error, any solution?